ArduPilot/MissionPlanner #896: Radio calibration, min max UI issue

github-0, 2015-06-07:
When doing the radio calibration today, I noticed the UI mixes throttle and yaw min/max red marks with each other. The dialog showing saved values is correct, so probably just a UI issue.

Tried to uninstall and reinstall MP, no change (the MP folder under program files didn't get removed when I used the uninstall wizard, so I deleted it manually). MP version 1.3.28, plane version 3.3.0.

[image]

andrea25, 2015-06-07:
Similar Issues here.
Look at my screenshot.

[image]

[image]

meee1, 2015-06-08:
this has been resolved in the current beta already. it is just a gui bug
https://github.com/diydrones/MissionPlanner/issues/883

closed 2015-06-08 by meee1