ArduPilot/MissionPlanner #908: Latest beta unable to perform update

lvale, 2015-06-19:
Several errors while copying files.

I have enable betas selected. Standard installation on Windows 7 -64

[image]
[image]

meee1, 2015-06-19:
please check missionplanner is not still running, ie in task manager. then retry.

lvale, 2015-06-20:
did that, and no it wasn't running

meee1, 2015-06-20:
the only reason it can fail is either file permissions, or the file is in use.
could be MP, antivirus, etc, etc

lvale, 2015-07-01:
Repeated the upgrades cycle and I believe it could have been a beta over beta issue, or networking snafu.....
[image]

closed 2015-07-01 by lvale