ArduPilot/MissionPlanner #911: CAM_TRIGG_TYPE resets to 0 when opening Camera Gimbal Setup

ThorstenDP, 2015-06-24:
That's pretty much it :-)
I use the Relay option to trigger the camera and also need to control tilt. It took me a while to find it out why my camera denied to trigger. But it is reproducible: 
- set CAM_TRIGG_TYPE to 1
- go to Initial Setup / Camera Gimbal (no need to click anywhere or change the settings)
- go back to the Full Parameter List and you have CAM_TRIGG_TYPE set to 0

Best regards,
Thorsten

paradisephill, 2015-07-07:
Yep, it's still an issue for me too. Basically, if you boot up Mission Planner and immediately change change CAM_Trigg_Type in the "Advanced Settings" window and then go to the Setup / Camera Gimbal window it changes CAM_Trigg_Type back to a default of "0".

airborneimager, 2015-07-11:
I think I can explain why different people get different behavior with this issue.

if the "Shutter" type in the Initial Setup/Camera Gimbal page is set to any RC channel, that setting "sticks" and will reset CAM_TRIGG_TYPE to 0 (unbeknownst to the user) any time this page is opened. On the other hand, if the "Shutter" type is set to "Relay" in the Initial Setup/Camera Gimbal page, it is NOT sticky - switching to the Full Parameter List shows the CAM_TRIGG_TYPE as 1, and going back to in the Initial Setup/Camera Gimbal page will show an empty/grayed out field for the Shutter type. This behavior has been present for as long as I can remember - more than a couple of years, so several dozens of versions of MP.

ThorstenDP, 2015-07-11:
I have to check, but it seems it only occurs since I use ch8 to control gimbal tilt.