ArduPilot/MissionPlanner #912: ARMING_CHECK checkboxes aren't displaying underlying value properly

rmackay9, 2015-06-27:
[image]

SkiTiSu, 2015-07-16:
This number is defined differently in values and bitmask. In  "Standard Params", this number is understood as bitmask but also write as a value. I'd like to know how to distinguish them? (Because a Param only have one value)

SkiTiSu, 2015-07-16:
This number is defined differently in values and bitmask. In "Standard Params", this number is understood as bitmask but also write as a value. I'd like to know how to distinguish them? (Because a Param only have one value)

<Values>0:Disabled, 1:Enabled, -3:Skip Baro, -5:Skip Compass, -9:Skip GPS, -17:Skip INS, -33:Skip Params/Sonar, -65:Skip RC, 127:Skip Voltage</Values>
<Bitmask>0:All,1:Baro,2:Compass,3:GPS,4:INS,5:Parameters+Sonar,6:RC,7:Voltage</Bitmask>

SkiTiSu, 2015-07-16:
@rmackay9 I have fixed this issue, but still have a problem. The "Values" of "ARMING_CHECK" are not the index of 2, will it cause some problems? Thx.

meee1, 2015-07-16:
the fix for this is here
db25e202

closed 2015-07-16 by meee1

SkiTiSu, 2015-07-16:
@meee1 db25e20 seems still have some problems:

For example, after I changed "ARMING_CHECK" to 4,
[image]
the checkboxes would not change in "Standard Params".
[image]

Everytime we entered "Standard Params", the List of checkboxes will be double counting
[image]

This time I only checked Baro, and click "Write Params" for many times, the value is accumulated into the max.
[image]
[image]

I think the problem is caused here,
```csharp
if (items[0].GetType() == typeof (MavlinkCheckBoxBitMask))
{
    ((MavlinkCheckBoxBitMask) items[0]).ValueChanged -= Control_ValueChanged;
    ((MavlinkCheckBoxBitMask) items[0]).setup(x.Key, MainV2.comPort.MAV.param); // here
    ((MavlinkCheckBoxBitMask) items[0]).ValueChanged += Control_ValueChanged;
    return;
}
```
maybe we could not setup it everytime.

In d2b3792, I fixed them.

Thanks.