ArduPilot/MissionPlanner #913: Heli Setup: not correctly displays the position angle of the servo at Swash-Servo position

ivanco070, 2015-06-29:
Dear developers, I apologize in advance for my English.
I use MP 1.3.30 build 1.1.5648.36304, I think, discovered a few bugs in MP.

1. Not correctly displays the position angle of the servo at Swash-Servo position. At the entrance to the Heli Setup it displays are always only the values 60, -60, 180, despite the fact that specified, for example, -120, 120, 0. If you put the cursor over the window manually entering the angle and then move it consistently on all the other servo, then the arrows at Swash-Servo position will move to correct position. But the next time you сome to Heli Setup it repeats 60, -60, 180 angles.
And some other problems with Heli Setup:
2. When I try to manually enter the Bottom of Swash Travel, I can't do it, once I have entered 4 digits, the value is changed for instance set by transmitter stick and nothing else.
You have to modify the Bottom in Full parameter list, but changing Bottom and making the write params, its impact on the swash plate immediately obvious, but Zero and Top, typed in Full parameter list will apply only after you restart pixhawk.
Interestingly, Zero and Top typed at Heli Setup impact on the swash plate immediately.
3. HSx_MIN and MAX settings are reset to initial (1000 and 2000) value when pixhawk restart.
But I do not now, problem 2 and 3 of MP or Pixhawk FW :)

I think the majority of users does not change the parameters HSx_MIN and MAX, because they have a normal servo. But I have a esky servo (t-rex clon not ordered yet :) ) and they have a non-linear stroke and different stroke length, which can be corrected for these parameters. And because, perhaps, no one previously has not encountered this problem :)

[image]

rmackay9, 2015-06-30:
I think those Trim, Min and Max fields should be removed because they can't be changed actually.  Instead the swash's roll-max and pitch max and swash travel should be used to limit the servo's movements.

ivanco070, 2015-06-30:
Randy, thank you very much for answering my post!

Trim, Min and Max are very helpful, when for the same band PWM control signal supplied to the swash servos, but the servos moves at different distances. In This case, to align the stroke swashplate helped me a lot these parameters. Also, I'm pretty sure, we may need to expand the control range of the PWM to the servo for some applications. In the case where the control only goes through pixhawk, and at the output of the transmitter we make the signal without CCPM and without ability to define separately for each servo CCPM band stroke, in the absence of these parameters, we completely lose the ability to flexibly manage the working stroke of the swashplate`s servos. For all other channels of the transmitter we have the opportunity to set the PWM to control the servo, But in the case of CPPM servo we will lose this opportunity. I think the lack of these parameters is very bad.
When we need to linearly reduce the whole band turn plates we can use Swah travel, in all other possible cases need the Trim settings, Min and Max. I would be happy for them :)
Especially in software of pixhawk and MP had laid, only to correct their work.

"Instead the swash''s roll-max and max pitch and swash travel should be used to limit the servo''s movements." - You mean physically on the helicopter with the reduction of the length of the servo arms? Or I just don't know about any settings allowing it to do?
To operate with these settings is also very convenient and they are very necessary. Leave it as is. Just don't delete them :)
But I think, like many others, will be happy if these settings will apply to work and immediately after entering, as it was originally intended, I guess.
Very sorry if too pushy in my opinion.