ArduPilot/MissionPlanner #916: Add support for nsh terminal over mavlink

rmackay9, 2015-07-01:
Tridge has recently added support for accessing the nsh terminal over MAVLink.  This was primarily for Solo which doesn't provide any access to the serial4/5 port on the PH2 but it would be useful for other developers who don't want to mess around trying to create a Debug Cable.
http://dev.ardupilot.com/wiki/apmpilot_project_developer_information/interfacing-with-pixhawk-using-the-nsh/

Details of how to do the setup with MAVProxy are at the bottom of the page above.  Tridge can probably provide details of the MAVLink messages to use to accomplish this.

meee1, 2015-07-01:
already done.
terminal > select nsh from the dropdown.
dont do it while in the air atm.....

meee1, 2015-07-01:
cf6afcbf421ae19a06e3634ed6c2d2263b0d3c8f

rmackay9, 2015-07-01:
ah, very cool.

closed 2015-07-01 by rmackay9