ArduPilot/MissionPlanner #922: simpleexample build error

williewonka, 2015-07-06:
hello i tried building the simple example with visual studio 2013 community edition.
However i get an error when building:
\ExtLibs\SimpleExample\simpleexample.cs(67,28,67,37): error CS0030: Cannot convert type 'byte[]' to 'T'
this corresponds to the line 67: 
return (T)packet;

How do i fix this?

PhilWheat, 2015-08-10:
I got it to work by changing that line to:
return (T)GetObjectFromBytes(packet, typeof(T));

And adding the following code:

       public static object GetObjectFromBytes(byte[] buffer, Type objType)
        {
            object obj = null;
            if ((buffer != null) && (buffer.Length > 0))
            {
                IntPtr ptrObj = IntPtr.Zero;
                try
                {
                    int objSize = Marshal.SizeOf(objType);
                    if (objSize > 0)
                    {
                        if (buffer.Length < objSize)
                            throw new Exception(String.Format("Buffer smaller than needed for creation of object of type {0}", objType));
                        ptrObj = Marshal.AllocHGlobal(objSize);
                        if (ptrObj != IntPtr.Zero)
                        {
                            Marshal.Copy(buffer, 0, ptrObj, objSize);
                            obj = Marshal.PtrToStructure(ptrObj, objType);
                        }
                        else
                            throw new Exception(String.Format("Couldn't allocate memory to create object of type {0}", objType));
                    }
                }
                finally
                {
                    if (ptrObj != IntPtr.Zero)
                        Marshal.FreeHGlobal(ptrObj);
                }
            }
            return obj;
        }

meee1, 2015-08-11:
fixed in master.

closed 2015-08-11 by meee1