ArduPilot/MissionPlanner #927: Mission planner mixing up units (meters and feet)

evanmacdonald9, 2015-07-10:
Hi there, I own an IRIS+ which I use with Mission Planner and have been having an issue with the altitude units. 

I planned a mission yesterday to fly at 90m. The IRIS rose to altitude and headed for its first waypoint. I quickly realized that it was certainly not at 90m and was barely above the tops of the trees which were about 70 ft at their highest. I brought the IRIS back and examined the plan to confirm that I was indeed using meters rather than feet, and looked in the configuration tab to confirm this was the case as well.

Do you have any suggestions on how to fix this issue? It's a good job I caught the altitude mix up or it could have been disastrous... 

Thanks in advance!

meee1, 2015-07-11:
please provide a tlog.

evanmacdonald9, 2015-07-12:
Hi, I'm new here so pardon my inexperience but I couldn't find out how to attach the tlog... It says only PNG, GIF or JPG are acceptable...

Thanks

meee1, 2015-07-13:
did you plan the mission in M or feet?

evanmacdonald9, 2015-07-13:
I did plan the mission in metres but I now know what was wrong... My first waypoint was quite far away and the IRIS was still climbing up to its set altitude of 90m. It was 30m (or about 90 ft after takeoff), then headed towards Waypoint 2, gradually ascending, but narrowly missing the tops of the trees.

I may change the takeoff alt to 50m. 

Thank you for the help, I will close this issue

closed 2015-07-13 by evanmacdonald9