ArduPilot/MissionPlanner #928: Altitude type bug

Jman841, 2015-07-12:
While I am still investigating the cause of this crash, I am almost positive it was caused by a bug in Mission planner.

Sometimes when loading mission planner the drop down menu under Waypoints in the Flight Plan can be blank. When this is blank I believe it defaults to Absolute altitude instead of Relative. After loading the mission to my plane today I did not realize this box was blank as it is normally defaulted to "Relative", but, since it was blank the plane defaulted to Absolute altitude and on take off it tried to go to the ground. I am still investing the cause of the crash, but, everything points to this error so far.

meee1, 2015-07-12:
please post the tlog,
as the tlog will contain what was actually uploaded.

Jman841, 2015-07-12:
I could not seem to figure out how to use the .tlog to find the absolute altitude of the waypoints, but, here is the file as I am sure it is only due to my lack of knowledge on diagnosing .tlogs.

http://ardupilot.com/forum/download/file.php?id=4193

Thank you

meee1, 2015-07-12:
just convert the entire log to a csv, then you can browse it as you require

meee1, 2015-07-12:
the files I have extracted shows a frame = 3, which is relative alt. so it was not MP.

also mp would have warned if you tried uploading a absolute alt mission, it assumes its a user error and prompts.

QGC WPL 110
1	0	3	16	0.000000	0.000000	0.000000	0.000000	4.823506	-74.155945	2586.000000	1
2	0	3	22	15.000000	0.000000	0.000000	0.000000	0.000000	0.000000	100.000000	1
3	0	3	178	0.000000	15.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
4	0	3	16	0.000000	0.000000	0.000000	0.000000	4.826026	-74.153481	150.000000	1
5	0	3	206	35.100000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
6	0	3	16	0.000000	0.000000	0.000000	0.000000	4.829645	-74.150154	150.000000	1
7	0	3	16	0.000000	0.000000	0.000000	0.000000	4.829175	-74.150963	150.000000	1
8	0	3	16	0.000000	0.000000	0.000000	0.000000	4.825556	-74.154297	150.000000	1
9	0	3	16	0.000000	0.000000	0.000000	0.000000	4.826416	-74.153885	150.000000	1
10	0	3	16	0.000000	0.000000	0.000000	0.000000	4.830035	-74.150558	150.000000	1
11	0	3	16	0.000000	0.000000	0.000000	0.000000	4.829566	-74.151375	150.000000	1
12	0	3	16	0.000000	0.000000	0.000000	0.000000	4.825945	-74.154701	150.000000	1
13	0	3	16	0.000000	0.000000	0.000000	0.000000	4.826805	-74.154297	150.000000	1
14	0	3	16	0.000000	0.000000	0.000000	0.000000	4.830425	-74.150963	150.000000	1
15	0	3	16	0.000000	0.000000	0.000000	0.000000	4.828784	-74.150558	150.000000	1
16	0	3	16	0.000000	0.000000	0.000000	0.000000	4.825167	-74.153885	150.000000	1
17	0	3	16	0.000000	0.000000	0.000000	0.000000	4.825637	-74.153069	150.000000	1
18	0	3	16	0.000000	0.000000	0.000000	0.000000	4.829254	-74.149742	150.000000	1
19	0	3	16	0.000000	0.000000	0.000000	0.000000	4.828394	-74.150154	150.000000	1
20	0	3	16	0.000000	0.000000	0.000000	0.000000	4.824778	-74.153481	150.000000	1
21	0	3	16	0.000000	0.000000	0.000000	0.000000	4.825249	-74.152664	150.000000	1
22	0	3	16	0.000000	0.000000	0.000000	0.000000	4.828864	-74.149338	150.000000	1
23	0	3	16	0.000000	0.000000	0.000000	0.000000	4.828003	-74.149750	150.000000	1
24	0	3	16	0.000000	0.000000	0.000000	0.000000	4.824389	-74.153069	150.000000	1
25	0	3	16	0.000000	0.000000	0.000000	0.000000	4.824860	-74.152252	150.000000	1
26	0	3	16	0.000000	0.000000	0.000000	0.000000	4.828473	-74.148933	150.000000	1
27	0	3	16	0.000000	0.000000	0.000000	0.000000	4.827016	-74.149887	150.000000	1
28	0	3	16	0.000000	0.000000	0.000000	0.000000	4.824000	-74.152664	150.000000	1
29	0	3	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
30	0	3	20	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
31	0	3	16	0.000000	0.000000	0.000000	0.000000	4.824967	-74.152863	100.000000	1
32	0	3	16	0.000000	0.000000	0.000000	0.000000	4.828281	-74.155396	70.000000	1
33	0	3	16	0.000000	0.000000	0.000000	0.000000	4.826164	-74.157928	30.000000	1
34	0	3	16	0.000000	0.000000	0.000000	0.000000	4.825063	-74.156921	15.000000	1
35	0	3	21	0.000000	0.000000	0.000000	0.000000	4.824048	-74.155998	0.000000	1

Jman841, 2015-07-12:
Ok,

If that is not the issue, I am quite stumped. Sorry for the confusion.

closed 2015-07-12 by Jman841

Jman841, 2015-07-12:
Do you see anything that would cause the issue that happened in the .tlog by chance?

The only thing that strikes me as Odd is the distance to WP on WP1 (Take off command) is over 8700 meters. no WP in the entire mission is 8700 meters away?