ArduPilot/MissionPlanner #929: Mission Planner issues: MAP zoom & SRTM graph

zenyo9, 2015-07-12:
Hi,

I use MP 1.3.30 and I've just downloaded last version.
I live in Reunion:
lat: -21,0991147676947
long: 55,251145362854

When I try to make a Flight Plan,I use the zoom and the resolution of the map is not enought good to work. I use GoogleSateliteMap ... but when I check GoogleEarth, the resolution is perfect... that's why I don t understand why it doesn t work on MP? 

My second issue is about the "graph elevation" when I make a Flight Plan, in Reunion, SRTM doesn t work.. the graph is ok for GE and MP but SRTM is totally FLAT...... about 0

If you guys could help me to fix that it would enable me to work on Mission Planner !!

Thx a lot !

meee1, 2015-07-13:
i just had a look, and there is srtm data for the island,
normaly MP will automaticlly download the correct data, please goto flightplanner, zoom in and move the mouse around, in the top left it should show you your mouse location with an altitude, it might take a little while to download, but should show the alt after a short time.

zenyo9, 2015-07-13:
Hi Michael,

Thx for help !!

About SRTM data, I still do not find anything :( 
I've tried with these WP:
* WP1 = (lat: -21,1059612 ) and (long: 55,2577972)
* WP2 = (lat: -21,0992349) and (long: 55,4069710)

[image]

On the WP2, Alt should be around 2000m - 2500m
When using MAP TOOL/Graph Elevation, I can see:
* blue line SRTM doesn t work: flat = 0
* green line GE works : until 2500m
[image]

You can also see on the WP1, when zoom in, the resolution is not enought good to create Flight Plan... that's why I can t work... 
[image]
But I don t understand because GOOGLE EARTH app resolution is very good on the same point....
[image]

zenyo9, 2015-07-13:
ok !
I just download the latest beta version, and the MAP Resolution zoom issue seems be fixed !!
I thinks you've done it? :D thx a lot.

About SRTM, I still do not get the good Alt..

=)

KiwiHC16, 2015-07-13:
Hi,
I have a similar issue SRTM at zero and 3Verify Height" which does work (except the zoom issue).
I use MP 1.3.30 build 1.1.5648.36304 with STIL and APM2.6 / V3.2.1. (same issue with MP beta) 
Issue 1: in Fligth Plan -> Map -> Elevation: SRTM values at 0. GE values look good.
Issue 2: there is no difference if I set or not "Verify Height" (perhaps I don't understand how it work). Alt are the altitude I fill in or defined by default. No change if GE alt at WP is higher (e.h. hill).
Thanks for your help

KiwiHC16, 2015-07-21:
Finally found the solution. MP use the SRTM to adjust the altitude and not the GE (Google Earth).
SRTM file should be in Mission Planner (MP) folder: mission planner/SRTM.
But you need to get the files yourself. For example, a mission to be planned at N64.5 W165.5 requires downloading the SRTM file N64W166.hgt.zip (identified by lower left corner of specific area), and unzipping it in the Mission Planner\srtm directory.
The file can be downloaded from http://dds.cr.usgs.gov/srtm/version2_1/SRTM3/, select they region and then select the file.
Now height-related functions (terrain and relative+verify) work.
Thanks for the information in this post: http://diydrones.com/forum/topics/use-google-elevation-for-verify-height
Shame that SRTM data are not downloaded by MP automatically. Is it a bug or is it not implemented ?

meee1, 2015-07-21:
it is automaticly downloaded,
try running MP as administrator.

KiwiHC16, 2015-07-21:
I just did the test with an other region and I had to download manual the SRTM file:
http://dds.cr.usgs.gov/srtm/version2_1/SRTM3/Eurasia/N42E001.hgt.zip

I’m the administrator on this computer.

Config:
Windows XP in a Virtual Machine (Parralel in OSX 10.10)
Mission Planner 1.3.30 build 1.1.5648.36304

Web browsing is fine from this PC.

In Flight Path tab, there is possibility to prefetch in the map menu but my understanding it for the map not SRTM, am I rigth ?

I’m not yet familiar with MP dev environment. Is there any specific test I can do to check what’s going on ? Any logs ?
If I have to install the dev environment, i will need some days before doing any investigation.

KiwiHC16, 2015-07-21:
I have now tried with an other PC and SRTM data are available.
They are available to check WP altitude and I can see the file in SRTM folder.
MP 1.3.30 build 1.1.5648.36304.
No VM.
Windows 7 Pro Service Pack 1.
I will redo the installation in the first PC to see.

KiwiHC16, 2015-07-21:
I removed mission planner and reinstalled it completely from fresh. Work well now. Was probably an installation issue.
Subject closed.

zenyo9, 2015-07-28:
Hi,
Ok, The MAP Resolution  still doesn t work again. I can get a really better resolution of the map by Google Earth than Mission Planner..:

[image]

the same place on Mission Planner:
[image]

About SRTM : Google Earth shows the right Alt but SRTM is FLAT...:
[image]

I use Windows 8.1
MP 1.3.30.3 build 1.1.5679.37990
I've tried old version, beta version, running in administrator.

Nothing works.. If you have an idea?
Can I expect it work one day in my area?

ty

zenyo9, 2015-07-29:
hello,

Do these issues happen only to me??? Do you think W8.1 can be the problem?

ty

zenyo9, 2015-07-29:
the main prob for me is the resolution of the map, because I can t work and can t use really MP with this kind of resolution:

[image]

another one:
[image]

Thx for interest.

bye

KiwiHC16, 2015-07-29:
I have exactly the same map behavior in 3 Mission Planner configuration (Blurd image) if I zoom too much over Reunion island.
I don’t have the issue in Google Earth.
But I have the same issue if you use Google map:
https://www.google.com/maps/place/21°12'10.4%22S+55°16'55.1%22E/@-21.2028876,55.2819854,14311m/data=!3m1!1e3!4m2!3m1!1s0x0:0x0 <https://www.google.com/maps/place/21%C2%B012'10.4%22S+55%C2%B016'55.1%22E/@-21.2028876,55.2819854,14311m/data=!3m1!1e3!4m2!3m1!1s0x0:0x0>

I see in one of your screen shot that SRTM values are at 0. I don’t have this issue. SRTM values look ok.

KiwiHC16, 2015-07-29:
One solution is to select an other map provider in the flight tab. I selected ArcGIS_Imagery_… and you can see nicely your area.

zenyo9, 2015-07-29:
thank s for your interest, I gonna try to uninstal et instal again MP and see if SRTM works... if not, I ll run MP with Windows 7......

About the map resolution issue, ArcGIS, BingSatelite etc. are too old (new road, and city was built since the satelite pictures have been taken.
Only Google Satelite is enought recent, but as you see, it doesn t works (and download ) enought to be able to work well.

My actual solution is:
* build the road flight with google earth
* import this file to MP
* write the survey on the drone and then fly

but I can follow my drone during the flight on Mission Planner, because the map resolution is too low..

if you get others idea to improve my working conditions, I will appreciate a lot !

ty

KiwiHC16, 2015-07-29:
Have you tried installing the SRTM file in MP folder manually ? It worked for me as a first solution (final solution was to do a fresh install of MP).

KiwiHC16, 2015-07-29:
I have exactly the same map behavior in 3 Mission Planner configuration (Blurd image) if I zoom too much over Reunion island.
I don’t have the issue in Google Earth.
But I have the same issue if you use Google map:
https://www.google.com/maps/place/21°12'10.4%22S+55°16'55.1%22E/@-21.2028876,55.2819854,14311m/data=!3m1!1e3!4m2!3m1!1s0x0:0x0

I see in one of your screen shot that SRTM values are at 0. I don’t have this issue. SRTM values look ok.

zenyo9, 2015-08-03:
Hey !

For information, I just instal Windows 10, and the SRTM fonction works perfectly now !

About the MAP the problem still appear in my Area in GoogleSat.. :'(