ArduPilot/MissionPlanner #932: Enhancement request: Set Loiter Radius button.

GrahamDyer, 2015-07-13:
One of the parameters that we find needing frequent changes during some missions is Loiter Radius in APM:Plane. A button to set the radius instantly would ease the process greatly as the only way to do change it currently without rewriting the mission is to search for it in the full parameter list using a keyword like "loiter" and then type in the change and write the parameter. Example shown in the image below. 

[image]

GrahamDyer, 2015-10-17:
http://diydrones.com/forum/topics/changing-loiter-radius-during-flight?xg_source=activity

GrahamDyer, 2016-02-22:
Bump, please. This would be so useful and time saving in APM:Plane missions.

GrahamDyer, 2016-04-01:
Bump again, was requested July 2015

meee1, 2016-04-01:
done

closed 2016-04-01 by meee1