ArduPilot/MissionPlanner #937: "This board has been retired, will upload latest version" when uploading AC3.1.5 to APM2

rmackay9, 2015-07-19:
The message is a bit misleading if the user has selected to upload a previous version of the firmware (like AC3.1.5) because "the latest version" is not uploaded but rather the AC3.1.5 they selected.

temi92, 2016-02-01:
HI, I am getting this sort of similar error as well in the process of trying to upload the firmware to the apm2.6 board.. along the line i get a "communication lost" error when uploading the firmware..how could i resolve this? Any advice would be greatly appreciated. thank you

temi92, 2016-02-01:
@rmackay9 HI, I am getting this sort of similar error as well in the process of trying to upload the firmware to the apm2.6 board.. along the line i get a "communication lost" error when uploading the firmware..how could i resolve this? Any advice would be greatly appreciated. thank you