ArduPilot/MissionPlanner #938: Bug in Log bitmask check box settings in Standard Params in beta

ThorstenDP, 2015-07-19:
Hi all,
the new nice log bitmask check box approach is not working properly in 1.3.30.2 1.1.5674.14075:
Each click is summed up, there is no reset resulting in wrong bitmask values.

Example:
- go to Full Parameter List
-    set log bitmask = 0

- go to Standard Params
-    check IMU
-    Write Params

- go to Full Parameter List
-   the bitmask value is now 128

- go back to Standard Parameters
-    uncheck IMU
-    check IMU
-    Write Params

- go to Full Parameter List again
-   the bitmask value is 256

- repeat...
-    the bitmask value constantly increases

Best regards,
Thorsten

ThorstenDP, 2015-07-19:
I made some more tests:
Basically it seems to work correct but once you made one wrong click and you want to uncheck a parameter you have to:
-> go to Full Parameter List
   --->     set the log bitmask value to 0

-> disconnect and close Mission Planner

-> Open Mission Planner  

-> go to Standard Params
   --->     check what you like to have logged
   --->     Write Params

-> disconnect and close Mission Planner

SkiTiSu, 2015-07-20:
@ThorstenDP I have solved it in d2b3792.

ThorstenDP, 2015-07-20:
@SkiTiSu thanks!

closed 2015-07-20 by ThorstenDP