ArduPilot/MissionPlanner #942: Request: User Item: Watts

iskess, 2015-07-21:
I'm starting to use Li-ion batteries which have much larger changes in voltage during the flight, from 4.2v to 2.7v. This makes Amperage and mAh used much less useful. I really need to see WATTS and Wh. 
Since the batt % is based off of mAh used, when the voltage varies greatly like it does with Li-ion, the % used is no longer linear and can be misleading. 
This would also help when switching between packs of different voltage.

meee1, 2015-07-29:
you have seen battery_usedmah?

iskess, 2015-07-29:
Yes, I use it all the time. Is there an easy way for me to multiply this by volts? 
What if you allowed for custom calculations using simple arithmetic like in Excel? This could be used for conversions too. Say I want all metric except altitude in feet to meet FAA requirements, I could make my own custom output.

iskess, 2015-08-02:
I'm liking this idea of allowing custom calculations for User Items. Here is another example:
I want to view my voltage divided by the number of cells. It easier to remember the significance of the values when switching between different cell counts.

iskess, 2015-08-02:
I changed the title of this issue to reflect the concept of custom calculations for User Items