ArduPilot/MissionPlanner #946: AC3.3 Geotagging with MP may not work

rmackay9, 2015-07-28:


taedla01, 2015-08-06:
Hello @rmackay9,

This mistake is due to the inclusion of the variable "Timeus" to CAM messages and restructuring of the message CAM data. Note that the variable "Timeus" has been included in all messages, so the messages ATT and GPS are also affected. To fix this, simply adjust properly "offtsets" in the setup screen Geotag Mission Planner tool.

The following configuration is incorrect (taken from the conversation on DIYDrones):
 
[image]

With new releases, for ArduCopter, Mission Planner must be set like this:

[image]

meee1, 2015-08-06:
thanks taedla01, ive updated the default to what you have said.

closed 2015-08-06 by meee1

taedla01, 2015-08-06:
Wooww! Very fast response @meee1!  ;)
I forgot to mention:

Tested with Mission Planner 1.3.30 and current master branch of ArduCopter (AC RC8+)

Thanks Michael,
Regards from Spain,
Dario.