ArduPilot/MissionPlanner #948: AC3.3 real-time vibration view

rmackay9, 2015-07-29:
[image]

It's becoming clear that high vibrations are extremely serious so the mock-up above attempts to make it easier for users to know their vibration levels in real time.

The "VIBE" field on the HUD should be Green when the highest vibration level (of the x,y,z values) is under 30.  If it's between 30 ~ 60 it should be amber/yellow.  Above 60 it should be red.

Also if the Clipping numbers have increased in the past second or so, the VIBE field on the HUD should definitely be red.

If the user clicks on the field the Vibration viewer should pop-up showing the individual X, Y and Z vibration levels and the total clip count.  Any non-zero Clipping values should be red... or possibly the clipping field should only go red for 1 second after the last time the clip count increased.

billbonney, 2015-07-29:
What MAVLink Message is this related too?

meee1, 2015-07-29:
mavlink_vibration_t

rmackay9, 2015-07-30:
Yes.  Here's the link to the MAVLink wiki page: http://mavlink.org/messages/common#VIBRATION

rmackay9, 2015-08-20:
We're getting closer to the AC3.3 release now so I just want to emphasize how important this issue and the EKF issue are.  My feeling is that they are show-stoppers for the release because it's important that users know the state of their vibration and EKF.

meee1, 2015-08-20:
these are in MP already, but only display with there is an issue. should they display all the time?

rmackay9, 2015-08-20:
@meee1, ah ok.
It would be good if the user can open it up themselves some how to check their vibration levels.