ArduPilot/MissionPlanner #950: Traditional Heli FF Tuning broken

Jman841, 2015-07-30:
Due to some changes in the parameter names (Now VFF and AFF for forward feed) the FF tuning parameter in mission planner no longer works under extended tuning for Arducopter Traditional Heli.

rmackay9, 2015-07-31:
AC3.3 supports VFF but AFF is being removed in the next release candidate.  So only the FF parameter needs to be update-able through the extended tuning page.

master continues to support AFF and it'll go out officially with AC3.4 most likely.

Jman841, 2015-07-31:
Ok, no issues, just noticed it didnt work.

Jman841, 2015-08-22:
So with AC 3.3 RC9 this VFF parameter is gone and it's back to FF so now it is broken again lol. Not sure what the plan is from Rob to call the parameters, maybe speak with him before changing it again as it may change again before the AC 3.3 stable release?