ArduPilot/MissionPlanner #952: Gimbal page resets camera trigger if there is more than one.

Jman841, 2015-07-31:
This is a rather annoying bug that had me going crazy for a little bit.

If you open the "Gimbal" Page under Initial setup and you have more than 1 RC channel as a camera trigger (Set to 10 for RC_Function) Mission planner will automatically set one of those RC Channels to 0 as soon as you click on the Gimbal page.

rmackay9, 2015-07-31:
I think this was added based on the assumption that we would only ever have a single camera attached.  I guess you're purposely trying to add more than one camera?  They would both receive the same pwm output so I suppose one solution we be to use a Y style servo cable to send the single pwm output from the flight controller to two servos.

I've thought a bit in the past about adding support for multiple cameras there would need to be a number of changes to do that including changing the mavlink do-camera-control messages to allow specifying an instance.

Jman841, 2015-07-31:
Yes, I have 2 drones with 2 camera's set up and it works perfectly. A Y cable will not work as one has the off command at 1500pwm and one has the off command at 1000 PWM. But both use 2000 PWM to trigger the camera

The trick to getting this to work is set the RC Servo minimum to 1500 on the camera where off is 1500 and it works perfectly.

Jman841, 2015-07-31:
But, I still don't know why Mission Planner would automatically set one of those to 0 just by clicking on the page. If 2 RC Outputs are set to Function 10, it shouldn't erase one of them.