ArduPilot/MissionPlanner #953: Wrong coordinate frame in waypoint files?

tengyifei, 2015-07-31:
In QGroundControl it's possible to specify the whether the altitude of a waypoint is relative to home or absolute. The coordinate frame of that waypoint is `MAV_FRAME_GLOBAL_RELATIVE_ALT(3)`  for relative and `MAV_FRAME_GLOBAL(0)` for the latter. Example:
```c
QGC WPL 120
/////// 0 means absolute
0	1	0	16	5	0	0	1	0	0	0	1
/////// 3 means relative
1	0	3	16	5	0	0	1	0	0	0	1
```
In Mission Planner however, waypoints are always relative to home, but the `MAV_FRAME` field in waypoints are always zero, incorrectly indicating that they are `MAV_FRAME_GLOBAL`. Is this the expected behavior?

Sample waypoint file from Mission Planner, version 1.1.30:
```c
QGC WPL 110
0	1	0	16	0	0	0	0	37.818586	-122.369016	20.000000	1
1	0	0	16	0.000000	0.000000	0.000000	0.000000	37.818226	-122.374692	30.000000	1
2	0	0	16	0.000000	0.000000	0.000000	0.000000	37.815209	-122.375636	30.000000	1
```

rmackay9, 2015-07-31:
Copter only supports relative altitudes at the moment.  Plane supports (only) absolute altitudes.
On the Copter side we hope to fix this in AC3.4 to allow relative, absolute or above terrain altitudes.  I think it'll be difficult for MP to make many improvements until this vehicle side work is done.

tengyifei, 2015-07-31:
What we could do at the moment is to fix the inconsistency. I suggest always using `MAV_FRAME_GLOBAL_RELATIVE_ALT` in the third column of the waypoint file to mark the waypoints as relative.

tengyifei, 2015-07-31:
After creating waypoints in Mission Planner and loading them in QGroundControl, QGroundControl prompts that the waypoints are not compatible with it. If you force QGroundControl to load it by changing the version tag from `110` to `120`, QGroundControl will load it but will treat all waypoint altitudes as absolute. This is a direct consequence of the MAVLink specs, where number zero in the third column means `MAV_FRAME_GLOBAL`. Given that MAVLink and QGroundControl are developed by the same team and whose behaviors do corroborate well, Mission Planner seems to be the odd one standing out here :frowning:

meee1, 2015-08-04:
I need more info on how you created the file.
as I tested and it worked as expected.

tengyifei, 2015-08-05:
Using Mission Planner 1.3.30 and QGroundControl 2.5.2

First issue:
1. Right click -> Draw polygon -> Add polygon point -> Finish drawing polygon
2. Right click on polygon -> Auto WP -> Survey (Grid)
3. Add takeoff and RTL
4. Save WP
Mission items are in `MAV_FRAME_GLOBAL`, but their altitudes are relative.

Second issue:
1. Rename generated waypoint file to `.waypoints` extension
2. Load generated waypoint file into QGroundControl
Gets this:
[image]

Full waypoint file:
```
QGC WPL 110
0	1	0	16	0	0	0	0	37.818586	-122.369016	20.000000	1
1	0	0	22	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	30.000000	1
2	0	0	16	0.000000	0.000000	0.000000	0.000000	37.815319	-122.369982	130.000000	1
3	0	0	206	28.096251	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
4	0	0	16	0.000000	0.000000	0.000000	0.000000	37.821734	-122.367381	130.000000	1
5	0	0	16	0.000000	0.000000	0.000000	0.000000	37.821590	-122.366555	130.000000	1
6	0	0	16	0.000000	0.000000	0.000000	0.000000	37.815226	-122.369134	130.000000	1
7	0	0	16	0.000000	0.000000	0.000000	0.000000	37.815133	-122.368287	130.000000	1
8	0	0	16	0.000000	0.000000	0.000000	0.000000	37.821445	-122.365728	130.000000	1
9	0	0	16	0.000000	0.000000	0.000000	0.000000	37.821301	-122.364901	130.000000	1
10	0	0	16	0.000000	0.000000	0.000000	0.000000	37.815040	-122.367439	130.000000	1
11	0	0	16	0.000000	0.000000	0.000000	0.000000	37.814947	-122.366591	130.000000	1
12	0	0	16	0.000000	0.000000	0.000000	0.000000	37.821156	-122.364074	130.000000	1
13	0	0	16	0.000000	0.000000	0.000000	0.000000	37.821011	-122.363247	130.000000	1
14	0	0	16	0.000000	0.000000	0.000000	0.000000	37.814854	-122.365744	130.000000	1
15	0	0	16	0.000000	0.000000	0.000000	0.000000	37.814761	-122.364896	130.000000	1
16	0	0	16	0.000000	0.000000	0.000000	0.000000	37.820867	-122.362420	130.000000	1
17	0	0	16	0.000000	0.000000	0.000000	0.000000	37.818232	-122.362603	130.000000	1
18	0	0	16	0.000000	0.000000	0.000000	0.000000	37.814668	-122.364048	130.000000	1
19	0	0	16	0.000000	0.000000	0.000000	0.000000	37.814575	-122.363200	130.000000	1
20	0	0	16	0.000000	0.000000	0.000000	0.000000	37.814936	-122.363054	130.000000	1
21	0	0	16	0.000000	0.000000	0.000000	0.000000	37.815454	-122.370813	130.000000	1
22	0	0	16	0.000000	0.000000	0.000000	0.000000	37.821879	-122.368208	130.000000	1
23	0	0	206	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
24	0	0	20	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
```

meee1, 2015-08-05:
what firmware had mp been connected to? also did you check what the altitude option box said in mp? I'm assuming it said absolute, so saved absolute

tengyifei, 2015-08-05:
It is not connected to any firmware. I couldn't find an option box that lets us specify the relative v.s. absolute altitude. :disappointed:

meee1, 2015-08-05:
MP changes the command available based on the MAV connected, as not all commands a supported on each vehicle type.
have you ever connected to a ardupilot vechile?

tengyifei, 2015-08-06:
Hmmm, after connecting to ardupilot the waypoints from AutoWP become relative, but there's still no option to explicitly specify whether to output absolute or relative.

If Mission Planner sees a waypoint file planned before connecting to an ardupilot (with absolute waypoints), does it automatically convert all waypoints to relative before sending them off to ardupilot?

meee1, 2015-08-06:
as randy said, this depends on the vehicle type.
copter only support relative.
plane support relative, absolute and terrain.

MP will ask when you goto upload if you wanted absolute, for copter this doesnt matter.

tengyifei, 2015-08-06:
If my understanding is correct, does this mean it will send `MAV_FRAME_GLOBAL` to ArduPlane and `MAV_FRAME_GLOBAL_RELATIVE_ALT` to ArduCopter?

meee1, 2015-08-06:
Relative = MAVLink.MAV_FRAME.GLOBAL_RELATIVE_ALT,
            Absolute = MAVLink.MAV_FRAME.GLOBAL,
            Terrain = MAVLink.MAV_FRAME.GLOBAL_TERRAIN_ALT

tengyifei, 2015-08-06:
Could you kindly clarify? I know that relative means `GLOBAL_RELATIVE_ALT` and absolute means `GLOBAL`. Are they being sent to ArduPilot and ArduPlane respectively no matter what's in the waypoint file?

meee1, 2015-08-06:
they are sent based on what is selected, for plane the menu shows, for copter it does not. The default for the drop down is relative.
look above the WP table in MP. there is a drop down that selects the alt mode, this decides how they are sent to the vehicle

tengyifei, 2015-08-07:
Ahh I see.. Is it ever possible to send ArduCopter `GLOBAL` waypoints (even if it might not make sense)? The drop down is hidden.

rmackay9, 2015-08-07:
For AC3.4 we plan to enhance Copter to support waypoints with altitudes specified as above-home, above-terrain or above-sea-level (i.e. absolute).  I don't think there's much point of changing mission planner until that work on copter is done because until then the requirements are unclear and it's difficult to test.

tengyifei, 2015-08-07:
Got it. I'm asking because we are using Mission Planner as the ground station for other UAV firmwares which support absolute altitude. Guess I'll have to stick to relative then.

closed 2015-08-07 by tengyifei