ArduPilot/MissionPlanner #955: Crash Info in HUD display during flight

kukulle, 2015-08-01:
I am flying a Hexacopter x-type with the new 3.3 beta software. I do not know if this issue is linked to the Pixhawk software or to the mission planner.
I recently was flying with the hex and I did some hard maneuvers (sudden climbs and hook-turns...) 
I suddenly received the information on the display (also acoustic message) that I crashed although I was still in the air.
I do not know where the measurement is done to give out information like this. I can only expect that there is a kind of g-load metering unit or the accelerometer who does this. But where are the limitations written. Onboard or in the Mission Planner?

rmackay9, 2015-08-01:
@kukulle, did the vehicle actually disarm in the air?
If you could put the logs on the AC3.3 beta testing thread that would be great.
http://diydrones.com/forum/topics/copter-3-3-beta-testing

kukulle, 2015-08-03:
Hello Randy,

as wished I've sent you the file.

Regards, Robert

meee1, 2015-08-07:
I'm going to close this, as I don't believe its a MP issue

closed 2015-08-07 by meee1