ArduPilot/MissionPlanner #958: Mission Planner using windows XP / Not showing flight modes

dudz996, 2015-08-04:
I have seen this issue all over the net, but no evidence of cures.

I have been Using versions of MP up until 2014 with great success along with my APM2.0 board.
I updated MP to the latest version, and loaded the latest possible firmware. Now, using my Turnigy 9x radio, on the MP Modes screen , I switch the radio to all my usual modes but MP does not show any modes, just blank tabs. Plus near the top of MP states "unknown". I have tried to load previous firmwares and MP versions, but it makes no difference. I just received a AMP2.6 clone, and it acts exactly the same with PM.  I am not using an OSD with the AMP so I don't
[image]
 know if the modes are showing there either.

meee1, 2015-08-07:
please do an update, on the help screen. this will redownload that missing file

Juemo, 2015-08-07:
Same probleme here with win XP for past week.
I have deinstall the MP and then I have use CCleaner tool for clean the Windows Registry
Then the new last MP installation works now ok here.

dudz996, 2015-08-08:
When I started MP thismorning, a new update automatically installed. This has indeed cured it. Thankyou !

  ----- Original Message ----- 
  From: Michael Oborne 
  To: diydrones/MissionPlanner 
  Cc: dudz996 
  Sent: Saturday, August 08, 2015 1:23 AM
  Subject: Re: [MissionPlanner] Mission Planner using windows XP / Not showing flight modes (#958)

  please do an update, on the help screen. this will redownload that missing file

  —

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

dudz996, 2015-08-08:
When I started MP this morning, a new update automatically installed. This has indeed cured it. Thankyou !

closed 2015-08-08 by dudz996