ArduPilot/MissionPlanner #960: Bug: LOITER_TO_ALT command will not upload to Plane, "Rejected"

iskess, 2015-08-06:
When attempting to upload a flight plan that contained LOITER_TO_ALT and DO_LAND_START to the plane I received an error, "Upload Failed, mission was rejected byt [real typo] the Mav, item had a bad option wp#39 MAV_MISSION_INVALID". (see photo below)
Interestingly WP#39 was the WP following the LOITER_TO_ALT command. 
After removing the LOITER_TO_ALT command, I still had an error. 
I then also removed DO_LAND_START and the upload was successful. 
I then add back only LOITER_TO_ALT and the upload failed again.
I then tried my friends set up with the same plane, same pixhawk FW 3.2, same version of MP 1.3.30. 
I uploaded a LOITER_TO ALT in his flight plan and it failed also 
I then removed it, and uploaded a DO_LAND_START which was successful.

[image]

Mission Planner 1.3.30, Plane 3.2, Pixhawk

meee1, 2015-08-06:
could you upload your wp file?

meee1, 2015-08-06:
seems like loiter_to_alt is not a valid command for the version of plane you are running.

iskess, 2015-08-06:
I pasted the flightplan below since I couldn't attach. I had compass problems with 3.3, I will try upgrading again after next weeks critical mission. 

QGC WPL 110
0	1	0	16	0	0	0	0	21.399756	-157.766247	68.939995	1
1	0	3	16	0.000000	0.000000	0.000000	0.000000	21.403531	-157.766017	117.000000	1
2	0	3	16	0.000000	0.000000	0.000000	0.000000	21.398676	-157.764514	117.000000	1
3	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399039	-157.764382	117.000000	1
4	0	3	16	0.000000	0.000000	0.000000	0.000000	21.403950	-157.765902	117.000000	1
5	0	3	16	0.000000	0.000000	0.000000	0.000000	21.403761	-157.765599	117.000000	1
6	0	3	16	0.000000	0.000000	0.000000	0.000000	21.398794	-157.764062	117.000000	1
7	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399157	-157.763929	117.000000	1
8	0	3	16	0.000000	0.000000	0.000000	0.000000	21.404179	-157.765484	117.000000	1
9	0	3	16	0.000000	0.000000	0.000000	0.000000	21.403990	-157.765181	117.000000	1
10	0	3	16	0.000000	0.000000	0.000000	0.000000	21.398913	-157.763609	117.000000	1
11	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399275	-157.763477	117.000000	1
12	0	3	16	0.000000	0.000000	0.000000	0.000000	21.404408	-157.765066	117.000000	1
13	0	3	16	0.000000	0.000000	0.000000	0.000000	21.404219	-157.764763	117.000000	1
14	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399031	-157.763157	117.000000	1
15	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399394	-157.763025	117.000000	1
16	0	3	16	0.000000	0.000000	0.000000	0.000000	21.404637	-157.764648	117.000000	1
17	0	3	16	0.000000	0.000000	0.000000	0.000000	21.404448	-157.764345	117.000000	1
18	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399149	-157.762705	117.000000	1
19	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399512	-157.762573	117.000000	1
20	0	3	16	0.000000	0.000000	0.000000	0.000000	21.404866	-157.764230	117.000000	1
21	0	3	16	0.000000	0.000000	0.000000	0.000000	21.404677	-157.763927	117.000000	1
22	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399267	-157.762253	117.000000	1
23	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399630	-157.762121	117.000000	1
24	0	3	16	0.000000	0.000000	0.000000	0.000000	21.405096	-157.763812	117.000000	1
25	0	3	16	0.000000	0.000000	0.000000	0.000000	21.404907	-157.763509	117.000000	1
26	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399386	-157.761801	117.000000	1
27	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399749	-157.761669	117.000000	1
28	0	3	16	0.000000	0.000000	0.000000	0.000000	21.405325	-157.763394	117.000000	1
29	0	3	16	0.000000	0.000000	0.000000	0.000000	21.405136	-157.763091	117.000000	1
30	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399504	-157.761348	117.000000	1
31	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399867	-157.761216	117.000000	1
32	0	3	16	0.000000	0.000000	0.000000	0.000000	21.405554	-157.762976	117.000000	1
33	0	3	16	0.000000	0.000000	0.000000	0.000000	21.405365	-157.762673	117.000000	1
34	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399622	-157.760896	117.000000	1
35	0	3	16	0.000000	0.000000	0.000000	0.000000	21.399985	-157.760764	117.000000	1
36	0	3	16	0.000000	0.000000	0.000000	0.000000	21.402333	-157.761505	117.000000	1
37	0	3	189	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
38	0	3	31	1.000000	-50.000000	0.000000	0.000000	21.401674	-157.763535	50.000000	1
39	0	3	16	0.000000	0.000000	0.000000	0.000000	21.402064	-157.764160	45.000000	1
40	0	3	16	0.000000	0.000000	0.000000	0.000000	21.401474	-157.766059	30.000000	1
41	0	3	16	0.000000	0.000000	0.000000	0.000000	21.400718	-157.765761	15.000000	1
42	0	3	21	0.000000	0.000000	0.000000	0.000000	21.400970	-157.764916	0.000000	1

closed 2016-03-31 by iskess