ArduPilot/MissionPlanner #963: float to double problem in params

magicrub, 2015-08-07:
params are transferred in float via MAVLink, but appear to be converted to doubles internally now. This happened from going tom 1.3.30 to 1.3.31. No big deal until you start interacting with param files.

Example: My file has EKF_VELD_NOISE,0.7 and EKF_WIND_PNOISE,0.1
When I go to compare to current settings I get
EKF_VELD_PNOISE (value) 0.699999988079071   (new) 0.7
EKF_WIND_PNOISE (value)  0.100000001490116   (new) 0.1

meee1, 2015-08-09:
agh yes, I fixed the log file saving/loading. just not the param compare

magicrub, 2015-08-19:
I see you've made a commit, is this fixed now? I can use beta but it's driving my team nuts. Can we push out a release soon?

magicrub, 2015-08-30:
fixed in the beta, thank you!

closed 2015-08-30 by magicrub