ArduPilot/MissionPlanner #965: Cannot change default SR0_xxx parameters in APM 2.7  from Mission Planner (to setup minimOSD)

dvillevald, 2015-08-09:
I cannot write SR0 parameters from Mission Planner to APM 2.7 (need this for minimOSD: https://code.google.com/p/minimosd-extra/wiki/APM) - whatever I type in (and try to Write) is lost after parameter refreshing. I tried MP 1.3.31 and 1.3.9, thru 3RD radio and USB port - nothing helps. 

P.S. No issues with SR1_xx parameters - they are loaded successfully
Thank you.

meee1, 2015-08-09:
goto the mp config page, and setup the stream rates for SR0 there. MP does this automatically

dvillevald, 2015-08-09:
Thank you meee1 for a prompt reply. I tried this (selected SR0_EXTRA1 as an example):
1. Connect to APM from MP
2. Go to Config/Tuning page -> Full Parameter list
3. Click on REFRESH PARAMS (which returns value 6 for SR0_EXTRA1) 
4. Change value of SR0_EXTRA1 from 6 to 5
5. Click on WRITE PARAMS
6. Click on REFRESH PARAMS 
7. The refreshed value of SR0_EXTRA1 is 6, not 5

Is it how you suggested to setup SR0 rates?

meee1, 2015-08-09:
mp update the values at a 60-90 second interval, and only if needed.

extra1 in mp forms part of the "attitude" option. ie 6

dvillevald, 2015-08-09:
If the values of SR0_xxx parameters are updated every 30 sec, how is it possible to set values of parameters SR0 to make minimOSD extra work with APM 2.7 (as stated in https://code.google.com/p/minimosd-extra/wiki/APM)?

In order for minimOSD extra to work, the link suggests:
"If you fly copter than you also need to set SR0 also:
SR0_EXT_STAT, 2
SR0_EXTRA1, 5 (I used it as example above)
SR0_EXTRA2, 2
SR0_EXTRA3, 3
SR0_POSITION, 2
SR0_RAW_SENS, 2
SR0_RC_CHAN, 5
Change accordingly and save the parameters to APM.'' 
Thank you again for helping.

meee1, 2015-08-09:
any number other than 0 means it is working.

meee1, 2015-08-09:
5 means 5 times per second
2 means 2 times per second

dvillevald, 2015-08-09:
I cannot make my minimOSD work as I cannot see layout over picture (I tried 2 minimOSD with different firmware/software, different wiring - nothing helps) so I assumed the SR0 rates could be the reason. Are you suggesting that actual values for SR0 rate are not that important (as long as they are not zero as you mentioned)? I read that these specific SR0 values make APM constantly send telemetry out. Thank you.

dvillevald, 2015-08-10:
It seems APM takes my SR0 parameters from https://code.google.com/p/minimosd-extra/wiki/APM but after mp update they are overwritten with SR0_EXTRA1=10  SR0_EXTRA2=10 SR0_EXTRA3=2 SR0_POSITION=3 SR0_RC_CHAN=2.

Are these values correct settings for SR0 params? Can user set values for SR0 parameters or are those set by mp (in latter case why they are called parameters)? Thank you again for helping.

meee1, 2015-08-11:
the mp values will work fine.

dvillevald, 2015-08-12:
Thank you again! As I cannot make minimODS-extra work with APM 2.7 I am hoping that recommended settings for SR0 parameters would help. BTW, others experience similar issue and provide suggestions how to make updated SR0 parameters "stick" (I have yet to try it):

http://www.rcgroups.com/forums/showthread.php?t=2274401&page=34&highlight=SR0#post32390877

dvillevald, 2015-08-15:
One more observation - I tried to write SR0 parameters from DroidPlanner (from tablet) and they do not stick either. Interesting however that I see different set of default values for SR0 params when I connect to DroidPlanner (2,10,10,...) compared to MP (2,6,6,...)

flashted, 2015-08-18:
Ok, I have the same problem, but I think I figured it out by accident as far as the "not updating" the SR0, and the SR1 parameters. 
  It kind of sucks, but it worked for me. I have been messing with this as others for quite some time, also. I could not update, so I decided to open a fresh brand new Mini APM 3.1 from my bins. I loaded APM Copter. I immediately went into the Advanced Parameters, to take a look. Guess what, some of the parameters (SR0) were set to zero's. I updated them to the described in the Wiki, and they stuck after writing. I went back, to make sure, and changed a few more, and they stuck... I thought my other board was bad, but wait! I tried it again, on the new board, and it DIDN'T work. 
   I then went and loaded APM PLANE, on the new board, and then went back and loaded APM Copter ( 3.2.1) to it again which resets the EEPROM, in a more simple manner BTW.  I then was able to set the SR0, and SR1 again. 
So that's it... 
 Have the values you want to change written down, do the above, change them, Viola!

Now.... Still No MAV Data  for me.  everything checks out. The cables are correct, and when I disable the telemetry, and plug in just the OSD, I have data, and overlay. Hmmm.

 I saw on the Wiki, there was a comment from a guy who suggested powering up the VTX, Camera, and the OSD separately either first, or second (don't remember) before, or after the telemetry is powered up by the main battery. I am suspecting this is the problem for me.

 Well my conclusion is, that if I cannot get them to work at the same time, I will just use the OSD for free flying, and the telemetry via my laptop for Missions. 
  
Plan B is to just set up a free standing OSD system. Sucks, but people who are not able to fix this, are doing it as a last resort.

 Here is the info... Hope it helps somebody

http://www.fpvstyle.com/plane/autopilot/fpv-osd-remzibi-hazy-firmware-apm-finished-products-set-band-gps.html

http://diydrones.com/profiles/blogs/port-minimosd-to-remzibi-s-hardware

https://code.google.com/p/hazys-osd/wiki/Remzibi

https://code.google.com/p/hazys-osd/

Another option...
Separate stand alone OSD (not hazy/ Remzibi)
http://www.thanksbuyer.com/qosd-module-full-function-osd-superposition-module-gps-module-for-qav-quadcopter-fpv-photography-40888

Cheers Ted

dvillevald, 2015-08-18:
Thank you for helping flashted! Did I understand you correctly that loading APM Plane first and APM Copter second helped to set SR0_xxx parameters? I will try it tomorrow.

flashted, 2015-08-18:
Hello Dimitri,  
Yes,  you heard it right. I figured that if I reset the EEPROM on the APM,  it would most likely work. 
I looked at doing it through Arduino, but it looked like a major pain, so, I found out that if you just load apm plane, and then re install  APM copter, and then immediately go back into the advanced parameters and look at SR0, and SR1, they most likely have changed to defaults. Then set what you need, and hit write params, and it should work. It did for me. BUT, like I said, make sure you have the params you want to set printed out, on paper, because it seems that you may have only 2 attempts to update,  before you are locked out again. This is exactly what happened to me.
 it seems that there is a major bug here, that has been found. But, the developers most likely are not interested in fixing it as all attention is on the Pixhawk now, hope this helps let me knowTed

flashted, 2015-08-19:
La!
 Got it all working. After I fixed the SR0 problem, I found that I had a cable problem also.

 It is very imperative that things are connected properly. GO BACK AND CHECK IT AGAIN......After many 3 AM nights working on this thing, I was fatigued. Brain not firing on all cylinders, You can make mistakes. Even a simple one can "Smokin Hole" your beautiful craft in seconds, with an average of $250-$450 worth of sleepless nights involved,  so it is a no brainier..

Now here is a description of how to hook your split cable to your OSD. (Not nescessarily in this order)

RX on OSD, and 3dr Telm are fed by the TX from your APM ONLY.

RX on APM fed by the Tx on 3dr only

Power to OSD, and 3dr Telm, is fed via APM by a APM power module fed by main battery. 
Or other means of 5.5 constant Volts.

Ground is common, hook them together

Here are my schematics for my APM 3.1 mini on is with a switch.

[image]
[image]
[image]

[image]

I will be posting my SR0, and SR1 parameters here in my next post.

Cheers!
Ted

dvillevald, 2015-08-19:
Ted,

Thanks a lot for sharing! Unfortunately loading APM plane and then APM copter did not work for me. SR1 params turned to 0s but SR0 were set to 2,6,6,1,... I tried to replace them with values from wiki and they did not stick :-( 

Resetting EEPROM from Arduino sketch did not help either. I will have to try something else...
Thank you again! 
Dmitri

flashted, 2015-08-19:
Ok, questions. ..
What version of copter are you using? I'm using 3,2,1 the very latest. 
What version of mission planner?  Get the very latest,  some believe the bug is not on the board.
I will have to check what version I have.
 If you are at a brick wall, get a new board, they are cheap.
This guy sells pre configured boards, at a great price. I buy all my stuff from him.I prefer the apm mini 3.1 buy it with his gps, basically plug and play.  
I will help you with the setup 
Where are you located? 
readytoflyquads.com

flashted, 2015-08-19:
I think you are out of options, aside from backing up in firmware versions. If you are in love with that board, and have the telemetry working, just go to a stand alone gps/ osd in the above posts. It is better than nothing.  I have a CC3D atom which doesn't support a gps, so I  may do so with that board. Good luck!

flashted, 2015-08-19:
One more thing, question..... did you load your parameters from a file before trying to write after the EEPROM was cleared? I tried that too, and it didn't work. I started right at the SR0/ SR1 parameters, changed them, hit write, then re configured everything else Through mission planner. 
 What kind of board do you have? 2.5, 2.6?
 My mini is a smaller version of the 2.6.
 DO NOT mess with the "purple " mini apms out there, I had 5 of them, and 3 of them had bad components on the with multiple warnings on power up. I dumped all of them in favor of RTF Quads version.  But, unfortunately I smoked one of them by disconnecting the power to the receiver,  and plugging it back in when connected via usb. A big no no lol

dvillevald, 2015-08-20:
Ted, 

Thanks again for helping, really appreciate it. I have the following setup:
APM Copter: version 3.2.1
MP: version 1.3.31 (tried 1.3.19 as well)
APM board: 2.7 (clone)
I tried two different APM boards with same results - SR0s do not stick - so I think board is okay. I am reluctant to replace it at this point as it quad flies fine and OSD is only thing I am missing. 

However, I made some progress. As it seems MP resets SR0 values to defaults, I installed Droid Planner on tablet and was able to set wiki parameters (!). Interesting - when I connect to MP, I still see old 2,6,6,1, when I connect to DroidPlanner, I see 2,5,2,3 - wiki minimOSD settings. It seems that both MP and DP reset values of SR0s to default ones, but in DP I can control those defaults while in MP I cannot. So for now I will be using DP to set wiki values.

Still no minimOSD layout however. Will try your wiring tomorrow and double check config and firmware on minimOSD.

Thanks a lot again for your help!
Dmitri 

P.S I am in Chicago, IL, what about you?

flashted, 2015-08-21:
Hey Dmitri, I'm in Sunny FL Almost all year flying LOL

Here is the latest version of Minim OSD from he developer...
I pulled this from DIY drones

I have started to add functions to MinimOsd code.
At first i did it for myself only. Added many functions i thought i need. Then opened this thread.
after a while, Pedro and later Miguel came, and things started to happen fast. :D
They have optimised the code and added even more things to it.
They have worked hard on CT, and it became a great tool!
Thank you Bough! :)
By now MinimOSD-Extra got a pretty advanced OSD.
Here it is in action:

Added:
- Changeable unit measurement (US, metric)
- Airspeed
- Home alt
- Battery Percent
- Battery used mah
- Current Draw
- Time From Startup (cleared at takeoff to show exact flight time)
- OSD Menu
- Variometer
- Wind horizontal speed and direction, and also the average wind speed of the last few minutes. 
- OSD on/off
- Switchable secound screen
- WP distance
- WP heading
- Crosstrack error
- Warning messages for Lost GPS fix, Stall, Overspeed, battery volt, battery Percent, RSSI
- Efficiency, glide distance & thermic notifier. 3 in one panel
- OSD Brightness
- HAM Call Sign
- After flight summary
- Trip distance
- Temperature
- Smoothened horizon
- Real heading
- RSSI
- Vertical speed
This functions can be turned on and off, and placed on different screens now, by the Config. tool.
Also RSSI, switching mode and channel and unit measurement, Stall speed warning, Overspeed warning, Battery warning volt, Battery percent warning, RSSI warning,  can be set in new Config Tool.
We built in a new way of setting video standards. Now OSD does not guessing anymore :). You can set it fixed from CT. It is in "Video Mode" menu.
Here is how it looks: (This video is a bit outdated, sorry. I will make a new one soon.)

The MinimOSD-Extra project is here: Link
This project is the developing version of the official Arducam OSD located here: Link
The latest stable version is: 2.2
The latest version can be downloaded from here: MinimOSD-Extra R800
Username: MinimOSD_Extra
Password: Top_Secret
CT is included. (The FW for Plane, Copter, Character upload and the character file is in the "FW & Char" directory inside CT directory)

flashted, 2015-08-21:
Link didn't post.

http://gabek.ddns.net/MinimOSD-Extra/MinimOSD-Extra_R800.zip

Username: MinimOSD_Extra
Password: Top_Secret

dvillevald, 2015-08-21:
SOLVED! At last it worked! I loaded version of minimOSD from your link above, wired the board - nothing. Then I (by accident) disconnected wires leading to camera and magically OSD layout appeared with no video. After re-connecting back camera wires I got both video and OSD working!

Here are my observations:
1. Using wiki-recommended SR0 values https://code.google.com/p/minimosd-extra/wiki/APM seems to help - after writing them to APM (from DroidPlanner as I could not make them stick from Mission Planner) OSD started to receive data. However, it might have something to do with a version of minimOSD-extra as I never tried R800 you suggested before.
2. Trick with disconnecting and then re-connecting back wires leading to fpv camera helped. I guess the order one powers APM and digital/analog parts of minimOSD matters. I power digital part of OSD from APM (<- power regulator <- battery) but analog part (camera and fpv Tx) - from balance plug of the same battery. So it seems the order should be: (1) power APM and digital part of minimOSD, then (2) power analog part (fpv camera and Tx) or minimOSD. I look forward to testing this.

Ted, thank you so much for your help, I really appreciate it - without it I would not be able to solve this puzzle! I am excited of playing with OSD screens and getting ready for my first FPV flight.

flashted, 2015-08-21:
Awesome Dimitri! , yes, I saw on another link, that there might be some sort of powering order for them to work. Also, check your camera wires again it sounds like you have some sort of bad connection there also, make sure that your ground wires are all connected Together. Vtx gnd, camera gnd osd gnd, etc. Especially if you are running only a primary battery. If you do not, you will get something called a floating ground. Check all of the wires on your OSD. Run a multimeter on all of the signal wires to confirm proper voltages and the VCC and ground to make sure everything is properly powered. you should get around 4.5 volts if you take your multimeter and put the ground to ground and then the positive to the signal wires like Rx Tx cts etc. With my OSD, I noticed that I could disconnect the power on the receiver transmitter side to the apm, and the board was still powered! Go figure... Also, move things around, and put pressure on the connectors while connected to see if anything shuts off. Vibration will crash you. I also experimented with connecting the Tx side of the OSD, to be connected to the Rx side of the apm. It doesn't seem to make a difference, but I left it unconnected for now because it works anyway. All of the wiring diagrams show it not to be connected. For some reason the pixhawk diagrams show it connected.
I also heard that you must try to change the parameters a few times, and then they will stick. That is what was reported on some forums. So it seems to be an intermittent thing. I am running a readytoflyquads.Com KV osd.
 Now, we are still not done. Now we have to get the extra features of these OSD's to come online. There is also a way to switch panels from your transmitter. I have some great links on this, that I will share with you. It's Friday night, which means a 3 a.m. stint in the garage stay tuned...

flashted, 2015-08-22:
[image]
[image]

Hello Dmitri
 My Beast,

I like Tri copters. I have my own design in the works.  It is a really cool concept. There will be nothing like it on the market, EXTREMlY fast, but if I produce it, the Chinesee will rip me off. Quickly. 

I also have a unibody Quad I plan to mold from a carbon kevlar weave, then race it. Very aerodynamic, and light. It seems like the racing quads are going aerodynamic. They are next on my plate.
 Right now I am fixing my strobes.  Just need to button up the wires, do final checks.....Then flight

Here is the link I told you about. I will tackle programming my OSD when I am done fixing my strobes.
https://code.google.com/p/minimosd-extra/wiki/Screen_Change

The above Tri has motors that will pull 59oz thrust at 14.8v with an 82% effency My experimental has the same motors. this is one of the best kept secret motors out there. $54 for 4

http://www.bevrc.com/bevrc-motor-elite-series-4006-kv650_p636.html
[image]
[image]

Hello Dmitri
 My Beast,

I like Tricopters. I have my own design in the works.  It is a really cool concept. There will be nothing like it on the market, Extremely fast, but if I produce it, the Chinese will rip me off. Quickly. I also have a uni body Quad I plan to mold from a carbon Kevlar weave, then race it. Very aerodynamic, and light. It seems like the racing quads are going aerodynamic. They are next on my plate.
 Right now I am fixing my strobes.  Just need to button up the wires, do final checks.....Then flight

Here is the link I told you about. I will tackle programming my OSD when I am done fixing my strobes.
https://code.google.com/p/minimosd-extra/wiki/Screen_Change

The above Tri has motors that will pull 59oz thrust at 14.8v with an 82% effiency My experimental has the same motors. this is one of the best kept secret motors out there. $54 for 4

http://www.bevrc.com/bevrc-motor-elite-series-4006-kv650_p636.html

Cheers

flashted, 2015-08-22:
http://www.bevrc.com/bevrc-motor-elite-series-4006-kv650_p636.html
[image]
[image]

Hello Dmitri
 My Beast,

I like Tricopters. I have my own design in the works.  It is a really cool concept. There will be nothing like it on the market, Extremely fast, but if I produce it, the Chinese will rip me off. Quickly. I also have a uni body Quad I plan to mold from a carbon Kevlar weave, then race it. Very aerodynamic, and light. It seems like the racing quads are going aerodynamic. They are next on my plate.
 Right now I am fixing my strobes.  Just need to button up the wires, do final checks.....Then flight

Here is the link I told you about. I will tackle programming my OSD when I am done fixing my strobes.
https://code.google.com/p/minimosd-extra/wiki/Screen_Change

The above Tri has motors that will pull 59oz thrust at 14.8v with an 82% effiency My experimental has the same motors. this is one of the best kept secret motors out there. $54 for 4

http://www.bevrc.com/bevrc-motor-elite-series-4006-kv650_p636.html

Cheers

flashted, 2015-08-22:
Triple post? Holy Crap what did I do?  Dont like that posts cannot be deleted....
Bummer

flashted, 2015-08-23:
This is from the link by Vortecks on RC Groups on how to solve this problem.

 First I must add, 
Get the R800 newest firmware, which comes with the newest configuration tool, in the upper part of this thread, and the newest version of Mission planner, and 3.2.1 copter firmware for your APM and install them.
http://www.rcgroups.com/forums/showthread.php?t=2026936

Quote:
"I was getting an error whilst trying to read from the MinimOSD.
Everytime I hit the button, I got “Failed to talk to bootloader” and then “EEPROM out of date” (or something like that).
After spending hours and hours, reading and learning, I finally found the answer and decided to share it with anyone else in the same situation......

Buy a usbasp programmer (around £5 from eBay). The drivers are available at:

http://www.fischl.de/usbasp/

Connect it up as per the attached images.

Run Arduino.

Select Tools > Board and select your board (mine is the Arduino Pro or Pro Mini (5v, 16MHz) w/ATmega128)

Select Tools > Programmer > USBasp

Select Tools > Burn Bootloader

Wait 2-3 mins and you’ll probably get an error of “warning: cannot set sck period. Please check for usbasp firmware update”. IGNORE IT!!!

Unplug the USBasp board and plug in your FTDI (with MinimOSD)

Open the MinimOSD configuration software, select the COM port and click ‘Read from OSD’

[image]

[image]

[image]

Now it takes a careful bit of timing, but hit the reset button on the MinimOSD then on the computer, hit "read OSD". Do this  through an FTDI cable to the 6-pin header on the Minim osd, and you should be able to upload firmware, charset, and do what you need to do.

Or, you can just connect a 4th wire from the DTS, on the FTDI loader to the GRN ( some versions say DTR) pin on the Minim OSD extra board, and not have to dick with hitting any buttons.

Use this method if you have to update the driver for your FT232RL FTDI USB Serial Adapter.

https://www.youtube.com/watch?v=RZH_qGautqM

[image]

Here is a video of the same thing in action.....

https://www.youtube.com/watch?v=FdYXnZZU7mY

I hope this info helps others who are struggling with this issue. I am going to surf some forums, to send them over here to get this info.

Cheers
Ted

dvillevald, 2015-08-23:
Great posts Ted, thank you so much! 

I am still working on adding minimOSD to my quad. When I power minimOSD from telemetry port of APM I can only get 4.4-4.5V which results in loss of fpv video signal once I switch on radio (at first I thought FrSky 2.4GHz signal interferes with fpv 5.8GHz but it does not seem to be the case as I completely lose video signal.) Powering minimOSD (digital part) from external 5V source (Arduino) resolves this issue so now I am trying to decide how to power minimOSD board (camera and fpv Tx are powered from balance plug of battery - I have one battery for everything.) Do you think it might help to power minimOSD from the same line which powers APM (i.e. straight from the output of power module)?  I read that power modules supplies up to 5.3V - is it safe for minimOSD board? (I have readytoflyquad KV OSD) Thank you again!

dvillevald, 2015-08-23:
To follow up on my previous post. I measured output off APM power module and got 5.3-5,4V. Description of KV OSD from readytoflyquads says that it has built-in 5V regulator which can handle up to 12V, so powering minimOSD from APM power mod output should be okay I think.

dvillevald, 2015-08-24:
Thank you for your posts Ted, it really helps! 

I am trying to figure out how to power digital part of minimOSD. Currently I power minimOSD from telemetry output of APM. However, it only gives me about 4.3-4.4V which does not seem to be enough as I lose video and OSD telemetry when I switch on my RC transmitter (I use FrSky 2.4GHz and fpv runs on 5.8GHz). When I provide independent 5V supply from external Arduino board to minim OSD then switching on RC does not affect video/OSD signal (12V power supply for fpv camera and fpv Tx are provided through the balance plug of the only battery I have on my quad.) What power options would you suggest for digital part of minimOSD?

flashted, 2015-08-24:
Hello Dmitri,

 Yes, it is safe. The newer boards do not have the voltage problems of the older boards.
 I power the APM, GPS, 3dr Telm, and Minim OSD from the power module only. Everything else is separate. 

 My Vtx, and camera are powered separately from a power distribution board which is fed by the main battery lines at 12 volts. 

In the picture you will see the XT60 plug / Pwr Module. The red and black wires from there, are going to the main distribution board at 12 Volts.
[image]

The video system power comes from the distribution board going through an LC filter (for noise) to the VTX, flight camera, which run at 12 Volts, and to a voltage step down converter to a Mobius which runs on 5.3 Volts. A separate system.

I also run my radio control system from the same distribution board through a step down voltage converter set at 6 Volts, and an LC filter. A separate system.

And last, I run my Strobe system from the same distribution board through a step down voltage converter set at 2.7 Volts for my killer 1 Watt low power consumption bright ass Led's (really cool)
I am still struggling to get the switch to work from my transmitter.  ;-(

LC filter

http://www.readymaderc.com/store/index.php?main_page=product_info&cPath=11_66_71&products_id=533

Step down converters
http://www.ebay.com/itm/331297126912?_trksid=p2055119.m1438.l2649&ssPageName=STRK%3AMEBIDX%3AIT

LED's
http://www.ebay.com/itm/111165784440?_trksid=p2055119.m1438.l2649&var=410211438161&ssPageName=STRK%3AMEBIDX%3AIT

[image]

Hope this helps.

I am now getting a No GPS Fix in MP, so I have to sit my Tri outside to let the GPS download its mapping for a half hour. Bummer. Almost ready to fly.

 I will keep you posted
Ted

dvillevald, 2015-08-24:
Update: I powered a digital part of minimOSD from analog A0 port of APM (+ to +5V and - to GND) and it works as it provides about 4.6V to minimOSD. Result: Video/OSD signal does not disappear anymore when I turn RC transmitter on. Bonus: no soldering is required.

I am still confused however that I can only get OSD telemetry when I do the following steps:
1. Power APM (and digital end of minimOSD)
2. Power analog part of minimOSD (fpv Tx and camera) thru balance plug of battery (video only is observed)
3. Unplug connector to camera (grey screen)
4. Push reset button on minimOSD (OSD telemetry shows up on grey screen)
5. Plug back camera (both OSD telemetry and video are present)

I wonder what might cause such behavior. I checked all connections to camera/vTx and they seem fine. All grounds have the same potential.

iskess, 2015-08-24:
I believe you are dealing with s camera compatibility problem. Try a different camera and you may find that the problem disappears. The typical SuperHAD Sony 600TVL is known to work well. http://www.securitycamera2000.com/products/600TVL-SONY-SUPER-HAD-CCD-D%252dWDR-Color-Board-Camera-with-OSD-Menu-DNR.html

flashted, 2015-08-24:
Ahh yes, check also that NTSC is checked in the CT tool for minim osd. If that is the format you wish to use, and if your camera is NTSC

dvillevald, 2015-08-25:
Iam, thank you for advice! I am using 480TVL Sony CCD FPV camera from Hobbyking Quanum FPV 5.8GHz Bundle Set.

Ted, thank you for helping again! The fpv set I use is PAL so I selected PAL in config tool for minimOSD.

flashted, 2015-08-25:
Hey Dmitri, 
Make sure you save the panel to your OSD. 
 You are only running 480 TVL? You should step up to a higher quality camera eventually.  You might see power lines better Lol. I do not go lower than 520 TVL, and pick the Wide Dynamic Range, Noise Reduction camers with the highest TVL. The link iskess posted above is an excellent camera,  and site. I have bought many camera's from them. The PZ0420 is a great choice,  along with a Sony Effio 960 H WDNR. Very CHEAP also. I am glad the prices have come down, as I have "smoked" quite a few learning this new aspect of the RC hobby. The camera s are getting really small now too.
 I have been busy, so no play time to start messing with the OSD panel switching, I suspect that my led strobes are not getting enough power, and the on off switch adjustment will be similar to the settings associated with the panel switching on the osd. Now we in S FL have a hurricane to worry about, and the preperation for that. I'm getting there though.

flashted, 2015-08-29:
Hello Dmitri,

Now I have a question for you....

 How do I reset the EEPROM for APM through Arduino?
I need to to try to clear a Mini 3.1 board, to determine if it is toast or not. I mentioned that I may have messed it it up in an earlier post. Need to know if I have to get a new one.

 I have Arduino installed on my computer to re set the Minim OSD boot loader through an AVR programmer, but if you can, could describe the process on an APM board through the mini USB connection?

Step, by step?
Cheers 
Ted

meee1, 2015-08-29:
I'm going to close this issue, as it has nothing to do with MP at this point in time.

closed 2015-08-29 by meee1

flashted, 2015-08-29:
Sorry, should have posted problem solved.
Being you are the man, any ideas as to why people are having this problem  with updating the SR0-SR1 parameters?  Is there a simple solution? 

I'm not worthy!  I'm  not worthy! 
He,He...

Hope you made a million before the Chinese ripped you off...
 I do have an account with 3dr, and have contributed to their account. ..
Cheers 
Ted

meee1, 2015-08-30:
if its plane, these values are saved between boots.
if its copter, they are not saved between reboots.
MP will always request data on the port it is connected to, the osd will as well if its setup with both rx and tx lines connected.

flashted, 2015-08-30:
Thank you Michael,

  Truly awesome technology.....

  My hat is off, and am bowed to you........

  You are truly a revolutionary.....

[image]