ArduPilot/MissionPlanner #974: Mission Planner update for AC3.3 - Autotune Flight Mode (+log) and Standard Params

andrea25, 2015-08-17:
There is some problems that I have seen in MP since using AC3.3 rc8.

I created this issues hoping that this can help your work.
If it is not needed please close it.

I know that the developers have a lot of work to do. I realy apreciate your work.
So, if these suggestion can cause undesired work, please ignore it.

1) Flight Mode - AutoTune
In AC 3.3 AutoTune is a Flight Mode.
Actually in the flight mode screen, AutoTune is not listed.
[image]

And also, if I go to full parameters list and I seek for FLTMODE1,2,3,4,5,6 I do not see Autotune in the suggestione.
The right number is 15 for AutoTune.
[image]

2) Log issue - The log do not read corretly the AutoTune flight mode

If I open a log of an AUto-Tune-Flight I do not see the AutoTune Flight Mode in the log.
I only see the AltHold Flight mode.
This is a screen of the issue.

Instead of display AutoTune, it display AltHold.
[image]

3) Autotune AGRR parameters

The parameter "AUTOTUNE_AGGR" is not listed in Standard Paras List.
Should it be listed?
Yes or no?

And also, in the Full parametrs List there is no description for the parameter "AUTOTUNE_AGGR"
[image]

4) Autotune Axes parameters

The parameter "AUTOTUNE_AXES" is not listed in Standard Params List.
Should it be listed?
Yes or no?

And also, in the Full parametrs List there is no description for the parameter "AUTOTUNE_AXES".
Maybe you could write something about 1 for roll, 2 for pitch and 4 for yaw.
[image]

5) ESC Calibration Starup Check parameters

The parameter "ESC_CALIBRATION" is not listed in Standard Paras List.
Should it be listed?
Yes or no?

And also, in the Full parametrs List there is no description for the parameter "ESC_CALIBRATION"
[image]

SkiTiSu, 2015-08-17:
@andrea25 It seems that your "ParameterMetaData" haven't updated. Press "Check for updates" in HELP Tab and it will automatically update "ParameterMetaData" from Github before update the main program.

andrea25, 2015-08-18:
@SkiTiSu Thank you for your reply. You are right! I did not notice that I have forgotten the beta update for MP.
Autotune flight mode is now ok, and is ok also in the log.
Sorry for posting without a real necessity.

But the descriptions of the parameters: ESC_CALIBRATION, AUTOTUNE_AGGR and AUTOTUNE_AXES are not upadted.
I hope this can help.

SkiTiSu, 2015-08-18:
@andrea25 Sorry, I forgot to remind that "Check for updates"  only get stale version of parameters. You can try "Check for beta updates" and it will get parameters from master branch. These parameters you method are in the master branch. ;)
[image]

andrea25, 2015-08-18:
@SkiTiSu Thank you for the reply!

I already have done the "Check for beta updates" but the descriptions of the parameters ESC_CALIBRATION, AUTOTUNE_AGGR and AUTOTUNE_AXES are not upadted. 

But it seems that it is only my problem.
So, now I go to reinstall the mission planner and to do another time the click on "Check for beta updates". 

Thank you again for reply.
Since it is only my issue, any moderator can close this issue. Sorry for opening and bothering.

closed 2015-09-24 by meee1