ArduPilot/MissionPlanner #976: Geotagging fails in MP 1.3.31

BangorSkis, 2015-08-19:
Original thread here: http://diydrones.com/forum/topics/geo-ref-not-working-after-update-of-mission-planner-1-3-31-1

I discovered today that using a BIN or LOG file from an Arducopter 3.2.1 mapping flight with CAM messages will result in:

     Error System.FormatException: Input string was not in a correct format.

Confirmed as other user had that using MP 1.3.29 works as expected.

meee1, 2015-08-19:
reduce all the numbers by 1

jmachuca77, 2015-08-19:
this is related to #973 I propose to read the APM version to decide on the offsets in order to allow both versions to work.

Jman841, 2015-08-28:
Is this fixed in the beta version? I need to do a mapping mission next week and need proper geotagging. Any updates, or, which older version is the geotagging working with?

Jman841, 2015-08-30:
Seems this is still not working correctly in the latest beta. However, Pix4d 2 now works properly with the flight logs again.

meee1, 2015-08-30:
realy? is the version your using not have any of the boxes at all anymore?

Jman841, 2015-08-30:
Unless I did something wrong. The cam messages mode wouldn't line up properly for some reason and kept telling me AMSL Alt can't be referenced. Then using the time offset method it gave me a very large number and only matched up about 1/4th the images. Also, you can no longer see the messages in the Geo ref images window. The only way to see the messages is to drag the window to the top of the screen and bring it into full screen mode, which, I luckily found out about on accident.

Maybe I am doing something wrong? I will try again tomorrow as I don't have those files on this computer. Pix4d geotagged all but 2 of the images perfectly using the log file from the flight.

CreativeBR, 2015-09-10:
Hello. The GPS offset -1 with Mission Planner geotag function worked to me but i prefer to use the Pix4D import 3DR log to geotag my photos because it have also attitude info.
Unfortunatelly it´s not working in Pix4D (Discovery 2.0.77). No photo is georeferenced.    
Any idea?

meee1, 2015-09-10:
mp does include attitude, look inside location.txt

CreativeBR, 2015-09-10:
Thank you meee1 but does Pix4D uses the location.txt? I guess it uses only the exif gps positions when not importing the 3DR log.

Jman841, 2015-09-10:
Yes, Pix4D Does use Location.txt. Simply choose "Lat, long, alt" from the parameter list and then load the location.

Also, please note, I asked Pix4D about the attitude information and they informed me that it is not used at all in their calculations. This was about 2 months ago, so it may have changed with the latest version.

CreativeBR, 2015-09-10:
Jman841 i think you are right... look at this: 
"Regarding Omega Phi Kappa - Yaw Pitch Roll - Camera Orientation
Updated: June 19, 2015 03:49
There is not a standard definition of Yaw, Pitch, Roll, and since the conversion to Omega, Phi, Kappa depends on several factors (position on the earth, output coordinate system), it is only possible to import them if the flight log belongs to any of our listed UAV manufacturers: 202558539.

The Omega, Phi, Kappa or the Yaw, Pitch, Roll values serve only for the calculation of the Geotag orientation variance (RMS) which appears in the Quality Report and the purpose is to compare the input values with the ones obtained by Pix4Dmapper, not for processing. 

Pix4Dmapper does not take into account the orientation angles for calibrating the images. Pix4Dmapper calculates the orientation angles during the first step (Initial Processing).

In the future we will use these values to speed up the processing. We do not believe that the results will be more accurate using these values. This is because, Pix4Dmapper calculates these values with very high accuracy due to the many (thousands) common characteristic points that finds on the images."

meee1, 2015-09-24:
going to close this, as 1.3.32 uses the fmt message to determine the field offset

closed 2015-09-24 by meee1