ArduPilot/MissionPlanner #978: Inject GPS doesn't allow BAUD rate choice for UDP Host

Jman841, 2015-08-26:
When I try to inject GPS for the Piksi RTK GPS's the latest version of mission planner has the BAUD rate for UDP Host blanked out and doesnt allow it to be selected or automatically select it. The only way to get it to work is to go to a COM port, choose the BAUD rate, then go back to UDP Host and it will be saved.

meee1, 2015-08-26:
the udp option does not use the baud rate. there was a bug in older versions that needed it set to something non 0, but that is fixed.

Jman841, 2015-08-26:
That is strange, if I click "Connect" with UDP Host selected and the baud rate blank it gives me an error that says "Invalid BaudRate" and does not let me continue. The only way to get it to work currently is to go to a COM port, choose a baud rate, then back to UDP Host.

meee1, 2015-08-26:
yes that was the bug I was talking about, the latest beta fix's that.

Jman841, 2015-08-26:
AH ok, I'll close this then.

Jman841, 2015-09-02:
Just tested again today with Beta 1.31.3 and it is still an issue. You still have to go to a COM port to choose the BAUD rate for UDP Host or else it says invalid baud rate.

meee1, 2015-09-24:
i discovered i only half fixed this.

meee1, 2015-09-24:
final fix is here
https://github.com/diydrones/MissionPlanner/commit/e51663df75c189a98bd1ad2b31a3f085190e208a

closed 2015-09-24 by meee1