ArduPilot/MissionPlanner #979: Mission Planner crashes with Log files

lvale, 2015-08-28:
While attempting to load a log file (120+ Mbytes )

[image]

Michael, if you want I can give you access to the original BIN files.

btw: it was impossible to retrieve the logs via the Mavlink interface, so I had to eject the card and copy the logs directly to the computer.

meee1, 2015-08-28:
are you sure the log you picked wasn't a blank file?

lvale, 2015-08-28:
Sure, even had Peter Barker confirm one of the files for me with MavTools

meee1, 2015-08-28:
I tested with a 800mb bin file and didn't have any issues. does mp do it everytime? do you have enough free disk space?

lvale, 2015-08-29:
400GB+ of disk free space. MP does it everytime. I'm on Mumble and send you the Google drive link to 4 recent logs that all do the same, 1.bin 2.bin 3.bin 4.bin each about 120MB each

meee1, 2015-08-29:
can you try the latest beta? as I cant reproduce

lvale, 2015-08-30:
Same issue with latest beta. Even on 10Mb Bin file, so it might not be related to file size but with some parameter

lvale, 2015-08-30:
Well. It turns out that Copter RC9 saved 130MB blank log files.....So not a MP issue.

closed 2015-08-30 by lvale