ArduPilot/MissionPlanner #980: Mission Planner latest update errors

lvale, 2015-08-29:
Today, MP pop'd up the new beta available and I accepted the offer to upgrade. 

Screenshot is part of the dialog that ends with, unable to upgrade. Tested on 2 different machines both with Windows 7 - 64

[image]

meee1, 2015-08-29:
restart windows, explorer.exe has that file locked. I see this one fairly often

closed 2015-08-29 by meee1

lvale, 2015-08-30:
Probably not
2 attempts, same result, both from fresh booted windows.
https://youtu.be/Ql9WXL3RQO4

meee1, 2015-08-30:
do you have tlogs on your desktop?

lvale, 2015-08-30:
Those tlogs are from Tower that I copied to the desktop. Unrelated issue.

meee1, 2015-08-30:
in this case they are related.
the tlog thumbnail driver will be trying to create thumbnails for them. and explorer.exe will be locking the dll because of it