ArduPilot/MissionPlanner #982: Bug in MP on Windows 10

Xerravon, 2015-08-31:
Hi,  I just got a new laptop formy ground station. I upgraded it to Windows 10 home x64. I want to try using a joystick so I went to set ti up.

Under actions tab MP is missing all but 3 buttons on any version higher than   MissionPlanner-1.2.99. so there is no joystick button.  on my win 7 pc it is working on any version.

the three buttons that show up on higher versions are "AUTO",  "MANUAL", and "RTL"

thanks,

John

meee1, 2015-08-31:
please goto mp options, and select advanced mode.

closed 2015-08-31 by meee1

Xerravon, 2015-08-31:
Hi,  I am not talking about the buttons along the top of the MP, I am talking about the buttons under the "actions" tab that are directly below the HUD on the "FLIGHT DATA"  page. where the "JOYSTICK" button should be.  these buttons are there on my windows 7 PC but only show up on the windows 10 PC in older versions of MP MissionPlanner-1.2.99 and below work fine but 	MissionPlanner-1.3.0 and above only show 3 buttons

If I understand you correctly your talking about the advanced view checkbox that adds or removes the simulation and terminal buttons. (along the top of the MP) If you mean something else I am not aware of any other MP options. where do I find them?

thanks,

John

Xerravon, 2015-08-31:
Hi,  I got it. Sorry I didnt look at the actions tab till 10 seconds after writing above. 

while I have you here, and I am not sure if I did something wrong or if it is a problem but you may want to look at it. I was running a beta ver of MP for a while, I uninstalled to go back to MissionPlanner-1.3.31. when I reinstalled MissionPlanner-1.3.31 I had a problem with the drivers for the radio modem. I tried uninstalling (and deleting leftover files and dir) a couple times. Finally I had to install an earlier version of MP then upgrade to  MissionPlanner-1.3.31 to get it to see my radio modem.

thanks for your help, sorry for my ignorance.

John