ArduPilot/MissionPlanner #984: Ability to Load old Beta Firmwares

Jman841, 2015-09-04:
Currently MP only supports loading old Stable Releases, however, I think it is more important to be able to load old beta firmwares, even if only for the current work being done.

With Beta firmwares, there is times when a new RC version will cause issues and currently it is not easy to revert back to an old Beta firmware.

This happened with RC6 on Arducopter and now RC10 for Traditional Heli. It would be great if I could revert back to RC9 from within mission planner.

meee1, 2015-09-06:
I'm not sure about this, there is a reason the beta was normally replaced. and don't think this would be a good idea.

Jman841, 2015-09-06:
The issue is sometimes the latest beta is unsafe to fly. This happened with RC6 on the current beta testing for Arducopter and is currently an issue with RC10 for Traditional Heli. I definitely understand that the new beta versions typically fix issues from previous iterations, but, being able to revert back at least 1 beta release could be good for people who test a new beta version and find something is broken. The old beta version might have a bug, but, maybe it is a minor issue vs. a new beta having a fatal issue?

rmackay9, 2015-09-07:
@Jman841, I don't think we can lay this problem at Michael's feet.  I think the solution is not to let bad release candidates hang out there as long as this one has.  We've been talking about allowing tradheli to do it's own releases which would free RobL to resolve this kind of thing without affecting the multicopter

Jman841, 2015-09-07:
That would work, and the RC6 issue was fixed rather quickly.

rmackay9, 2015-09-07:
Ok great.  New issue in the ardupilot build env to add the tradheli build tag support and it seems like Tridge and Grant will implement his today.
https://github.com/diydrones/ardupilot/issues/2809

I'm going to close this issue as the other one is what we all agree we can live with.

closed 2015-09-07 by rmackay9