ArduPilot/MissionPlanner #985: Support for custom terrain data

rmackay9, 2015-09-06:
This is a request that came up on the original add-support-for-terrain-data issue.

ReMapHI, 2015-09-06:
Thanks @rmackay9. I just wanted to point out that this issue has been recently brought up on the Ardupilot forum as well:
http://ardupilot.com/forum/viewtopic.php?t=13628

meee1, 2015-09-06:
what I plan on doing is adding geotiff support, but only wgs84 geotiffs, aka geographic geotiffs

ReMapHI, 2015-09-06:
@meee1 That should work for us (and other GIS folks)

meee1, 2015-09-06:
this should be in the latest beta.
you need to copy your geotiff's to the srtm directory in the mp directory

ReMapHI, 2015-09-06:
Thanks. Once the geotiffs are in the srtm directory, is there any setting within mp that needs to be adjusted in order to read from those geotiffs, or will the elevation be automatically read from them instead of the srtm data?

meee1, 2015-09-06:
by default it will use the geotiff, and fall back to srtm.

meee1, 2015-09-15:
everybody happy with this?

ReMapHI, 2015-09-15:
@meee1 We'll need some time to test this. We'll let you know soon.

ReMapHI, 2015-09-17:
@meee1 I tried making a survey grid plan twice--with and without the custom geotiff--and checking the "verify height" option. Upon examining the altitude values in both WP files, it appears that they are identical. Also, when I tried to use the elevation graph map tool, MP said it was using SRTM, even when I had the custom geotiff in the folder.

meee1, 2015-09-17:
srtm and the geotiff are inter changeable., if there is no geotiff data for the area it will use srtm.
I did find a bug in the geotiff parser to do with the start image cords, and the half pixel offset.
https://github.com/diydrones/MissionPlanner/commit/bb221e191046e7f13ce70abf8a1e8776aa3aca93

meee1, 2015-09-17:
please post your missionplanner.log file, this should have the basic image info in it

ReMapHI, 2015-09-17:
It appears that the geotiff is recognized. See log below.

It's highly unlikely that the srtm and geotiff data are identical in this area. We can try to test with a different area.

-----

2015-09-17 06:41:58,468  INFO Program - ******************* Logging Configured ******************* (c:\Users\michael\Source\Repos\MissionPlanner\Program.cs:47) [1]
2015-09-17 06:41:58,844  INFO MissionPlanner.MainV2 - Mainv2 ctor (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:314) [Base Thread]
2015-09-17 06:41:58,989  INFO MissionPlanner.Utilities.Speech - TTS: init, mono = False (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\Speech.cs:43) [Base Thread]
2015-09-17 06:41:59,013  INFO MissionPlanner.MainV2 - Loading config (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:1475) [Base Thread]
2015-09-17 06:41:59,019  INFO MissionPlanner.MainV2 - Create FD (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:503) [Base Thread]
2015-09-17 06:41:59,026  INFO MissionPlanner.GCSViews.FlightData - Ctor Start (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:165) [Base Thread]
2015-09-17 06:42:00,006  INFO MissionPlanner.GCSViews.FlightData - Components Done (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:169) [Base Thread]
2015-09-17 06:42:00,007  INFO MissionPlanner.GCSViews.FlightData - Tunning Graph Settings (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:184) [Base Thread]
2015-09-17 06:42:00,022  INFO MissionPlanner.GCSViews.FlightData - HUD Settings (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:226) [Base Thread]
2015-09-17 06:42:00,029  INFO MissionPlanner.Common - getModesList Called (c:\Users\michael\Source\Repos\MissionPlanner\Common.cs:652) [Base Thread]
2015-09-17 06:42:00,082  INFO MissionPlanner.GCSViews.FlightData - Graph Setup (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:270) [Base Thread]
2015-09-17 06:42:00,083  INFO MissionPlanner.GCSViews.FlightData - Map Setup (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:274) [Base Thread]
2015-09-17 06:42:00,085  INFO MissionPlanner.MainV2 - Create FP (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:505) [Base Thread]
2015-09-17 06:42:00,345  INFO MissionPlanner.MainV2 - Create SIM (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:508) [Base Thread]
2015-09-17 06:42:00,457  INFO MissionPlanner.MainV2 - Saving config (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:1436) [Base Thread]
2015-09-17 06:42:00,474  INFO MissionPlanner.MainV2 - myview width 1301 height 745 (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2404) [Base Thread]
2015-09-17 06:42:00,482  INFO MissionPlanner.MainV2 - this   width 1317 height 784 (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2406) [Base Thread]
2015-09-17 06:42:02,887  INFO MissionPlanner.MainV2 - Load Pluggins (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2137) [Base Thread]
2015-09-17 06:42:02,974  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\MissionPlanner.Grid.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:49) [Base Thread]
2015-09-17 06:42:02,979  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Grid 0.1 by Michael Oborne (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:60) [Base Thread]
2015-09-17 06:42:02,989  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\MissionPlanner.Gridv2.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:49) [Base Thread]
2015-09-17 06:42:02,992  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Grid 0.1 by Michael Oborne (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:60) [Base Thread]
2015-09-17 06:42:03,001  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\MissionPlanner.SimpleGrid.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:49) [Base Thread]
2015-09-17 06:42:03,004  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init SimpleGrid 0.1 by Michael Oborne (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:60) [Base Thread]
2015-09-17 06:42:03,011  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\MissionPlanner.Stats.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:49) [Base Thread]
2015-09-17 06:42:03,078  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init Stats 0.1 by Michael Oborne (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:60) [Base Thread]
2015-09-17 06:42:03,097  INFO MissionPlanner.Plugin.PluginLoader - Plugin Load C:\Program Files (x86)\Mission Planner\plugins\TrackerHome.dll (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:49) [Base Thread]
2015-09-17 06:42:03,099  INFO MissionPlanner.Plugin.PluginLoader - Plugin Init GPS Tracker Home Plugin 1.0 by Will Bryan (c:\Users\michael\Source\Repos\MissionPlanner\Plugin\PluginLoader.cs:60) [Base Thread]
2015-09-17 06:42:03,130  INFO MissionPlanner.MainV2 - Load Pluggins Done (c:\Users\michael\Source\Repos\MissionPlanner\MainV2.cs:2139) [Base Thread]
2015-09-17 06:42:03,143  INFO MissionPlanner.GCSViews.FlightData - Activate Called (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightData.cs:491) [Base Thread]
[... cut ...]
2015-09-17 06:43:55,106  INFO MissionPlanner.GCSViews.FlightPlanner - param LOITER_RADIUS 45 (c:\Users\michael\Source\Repos\MissionPlanner\GCSViews\FlightPlanner.cs:2109) [Base Thread]
2015-09-17 06:43:55,455 DEBUG MissionPlanner.Utilities.Tracking - OK (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\Tracking.cs:288) [26]
2015-09-17 06:43:55,579  INFO MissionPlanner.Utilities.GeoTiff - GeoTiff C:\Program Files (x86)\Mission Planner\srtm\4QFJ622370_lmsl_wgs84.tif (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:29) [Base Thread]

meee1, 2015-09-17:
looks like you geotiff doesn't have the proper geotiff tags
it should have spat out

2015-09-17 06:35:37,066  INFO MissionPlanner.Utilities.GeoTiff - GeoTiff C:\Users\michael\Source\Repos\MissionPlanner\bin\Debug\srtm\srtm_59_19.tif (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:29) [Base Thread]
2015-09-17 06:35:37,106  INFO MissionPlanner.Utilities.GeoTiff - Tie Point (0,0,0) --> (109.999583236466,-29.9995833333434,0) (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:50) [Base Thread]
2015-09-17 06:35:37,106  INFO MissionPlanner.Utilities.GeoTiff - Scale (0.000833333333333333,0.000833333333333333,0) (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:56) [Base Thread]
2015-09-17 06:35:37,107  INFO MissionPlanner.Utilities.GeoTiff - Coverage {Lat=-29.9995833333434,Lng=109.999583236466,WidthLng=5.00083333333333,HeightLat=5.00083333333333} (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:60) [Base Thread]
2015-09-17 06:35:50,451  INFO MissionPlanner.Utilities.GeoTiff - GeoTiff C:\Users\michael\Source\Repos\MissionPlanner\bin\Debug\srtm\srtm_60_19.tif (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:29) [Base Thread]
2015-09-17 06:35:50,461  INFO MissionPlanner.Utilities.GeoTiff - Tie Point (0,0,0) --> (114.999583357538,-29.9995833333434,0) (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:50) [Base Thread]
2015-09-17 06:35:50,461  INFO MissionPlanner.Utilities.GeoTiff - Scale (0.000833333333333333,0.000833333333333333,0) (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:56) [Base Thread]
2015-09-17 06:35:50,462  INFO MissionPlanner.Utilities.GeoTiff - Coverage {Lat=-29.9995833333434,Lng=114.999583357538,WidthLng=5.00083333333333,HeightLat=5.00083333333333} (c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Utilities\GeoTiff.cs:60) [Base Thread]

ReMapHI, 2015-09-17:
Thanks @meee1 . So how do we get proper geotiff tags? Are tfw or prj files required?

The original elevation data was in ERDAS Imagine img format, which I converted/reprojected to geotiff using Global Mapper.

meee1, 2015-09-17:
no, for a geotiff technically the tfw and prj are not needed, as the geotiff should have tags with the same info, I'm not sure how globalmapper does this though.

meee1, 2015-09-24:
any luck with this one? let me know as i may need to add tfw support.

ReMapHI, 2015-09-25:
I'm not really sure how to proceed, since I don't understand how Global Mapper converts/reprojects the geotiff either. Adding tfw support might be helpful. Is that difficult to implement?

vfp1, 2016-07-21:
Hi, might be a little to late for this but this worked for me without use of .tfw files. I used QGIS. Import the raster with the elevation information into QGIS. Right click the raster layer in Layers Panel and click "Save As...". The GTiff format is preselected and you can modify the final extents of your geotiff according to your needs. Make sure that the CRS (Coordinate Reference System) is set as WGS 84 (EPSG: 4326). Click OK and you will have a .tiff file with geotags. 

Then I just copied that file into the SRTM folder. So I believe that when we select "Terrain" in the waypoints tab in Flight Plan, MissionPlanner will use by default the GeoTiff from the area in which the mission is planned, isn't it?

ReMapHI, 2016-07-22:
I downloaded the USGS 10m DEM for Oahu and compared it to SRTM by clicking on few locations in GIS and noting the differences in elevation between datasets. Then I reprojected the USGS DEM into WGS84 and copied the GeoTIFF (just the *.tif) into Mission Planner's SRTM folder. I opened Mission Planner and checked the same spots by hovering the cursor as well as creating waypoints with the different options (absolute, relative, terrain). The elevation data is still coming from SRTM, not the USGS DEM that I copied. 

Is there something that we are missing @meee1 ?

To answer the question from @vfp1 : the "Terrain" option will ignore elevation data and only use the default alt (the same result as leaving the "Verify Height" box unchecked).

imptah, 2021-08-19:
@meee1 can you please help me, I put my GeoTIFF file inside srtm folder, but it was doesn't used by MP. In the log, I see that geotiff file was recognized but there is an exception after it:

```
2021-08-19 21:14:07,546  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=GTModelTypeGeoKey TagLoc=0 Count=1 Value/offset=1 (:0) [25]
2021-08-19 21:14:07,547  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=GTRasterTypeGeoKey TagLoc=0 Count=1 Value/offset=1 (:0) [25]
2021-08-19 21:14:07,547  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=GTCitationGeoKey TagLoc=34737 Count=35 Value/offset=0 (:0) [25]
2021-08-19 21:14:07,554  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=GTCitationGeoKey Value=Estonian Coordinate System of 1997| (:0) [25]
2021-08-19 21:14:07,554  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=GeogCitationGeoKey TagLoc=34737 Count=6 Value/offset=35 (:0) [25]
2021-08-19 21:14:07,554  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=GeogCitationGeoKey Value=EST97| (:0) [25]
2021-08-19 21:14:07,554  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=GeogAngularUnitsGeoKey TagLoc=0 Count=1 Value/offset=9102 (:0) [25]
2021-08-19 21:14:07,555  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=2062 TagLoc=34736 Count=3 Value/offset=0 (:0) [25]
2021-08-19 21:14:07,555  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=ProjectedCSTypeGeoKey TagLoc=0 Count=1 Value/offset=3301 (:0) [25]
2021-08-19 21:14:07,555  INFO MissionPlanner.Utilities.GeoTiff - GeoKeyDirectoryTag ID=ProjLinearUnitsGeoKey TagLoc=0 Count=1 Value/offset=9001 (:0) [25]
2021-08-19 21:14:07,555  INFO MissionPlanner.Utilities.GeoTiff - GeoAsciiParamsTag Estonian Coordinate System of 1997|EST97|  (:0) [25]
2021-08-19 21:14:07,556  INFO MissionPlanner.Utilities.GeoTiff - Tie Point (0,0,0) --> (365000,6635000,0) (:0) [25]
2021-08-19 21:14:07,556  INFO MissionPlanner.Utilities.GeoTiff - Scale (25,25,0) (:0) [25]
2021-08-19 21:14:07,557  INFO MissionPlanner.Utilities.GeoTiff - Coverage {Lat=0,Lng=0,WidthLng=0,HeightLat=0} (:0) [25]
2021-08-19 21:14:07,557  INFO MissionPlanner.Utilities.GeoTiff - Start Point (0,0,0) --> (365012.5,6634987.5,0) (:0) [25]
2021-08-19 21:14:07,807 DEBUG MissionPlanner.Program - FirstChanceException in: Microsoft.CodeAnalysis (:0) [4]
System.DllNotFoundException: Unable to load DLL 'Microsoft.DiaSymReader.Native.amd64.dll': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at Microsoft.DiaSymReader.SymUnmanagedFactory.CreateSymWriter64(Guid& id, Object& symWriter)
   at Microsoft.DiaSymReader.SymUnmanagedFactory.CreateObject(Boolean createReader, Boolean useAlternativeLoadPath, Boolean useComRegistry, String& moduleName, Exception& loadException)
```

Can you please comment on it?