ArduPilot/MissionPlanner #986: Failed to update home location when SToRM32 gimbal is attached using MAVLink protocol

rmackay9, 2015-09-06:
When using Copter-3.3 with a SToRM32 gimbal and the MAVLink protocol, soon after connecting and at the end of downloading parametes (from perhaps both the vehicle and the gimbal) a "Failed to update home location" message is displayed to the user.
My guess is that Mission Planner is either trying to retrieve or perhaps set the gimbal's home location which is not possible and this message appears.
At the time this happens the vehicle selector box is also on the screen but the user can't modify it because the parameter download is ongoing with a modal window.

meee1, 2015-09-06:
do you have arming turned off?

rmackay9, 2015-09-06:
@meee1, Actually it's not possible to disable arming in copter.  the ARMING_CHECK parameter is also "1" in case that's what you meant.

closed 2015-09-24 by meee1