ArduPilot/MissionPlanner #989: MP latest beta missing descriptions on full parameter list

imrj, 2015-09-06:
hi, since about two ot three beta releases ago I noticed that a lot of the full parameter list descriptions have gone missing completely, which were previously there....some of these are below (not inclusive of all that are missing)

ATC_ACCEL_P_MAX
ATC_ACCEL_R_MAX
AUTOTUNE_AGGR
FS_EKF_ACTION

rmackay9, 2015-09-07:
maybe the beta firmware is no longer pointing at ardupilot/master for the parameter info?
So I think the issue is somewhere in the MissionPlanner as opposed to the ardupilot code.
i'm often wrong of course but that's my guess.

meee1, 2015-09-07:
beta mp downloads params from master.
release mp downloads from stable. 
this was to prevent param creep on stable builds

imrj, 2015-09-08:
so...what does this mean?  :)

meee1, 2015-09-24:
if you need to click the beta button in mp to upload the firmware you are running, then you need to update to beta MP to get the correct documentation.

closed 2015-09-24 by meee1