ArduPilot/MissionPlanner #990: support for on-board compass calibrator

tridge, 2015-09-07:
We have the new on-board compass calibrator in master now and it will also be in the 3.4.0 plane release. You can see the protocol here:
https://github.com/diydrones/ardupilot/blob/master/libraries/GCS_MAVLink/message_definitions/ardupilotmega.xml#L713
and an implementation in MAVProxy here:
https://github.com/DroneCode/MAVProxy/blob/master/MAVProxy/modules/mavproxy_calibration.py#L115
the on-board mag cal can do soft-iron corrections which can be very worthwhile on some aircraft.
It would be great to get support in MissionPlanner for this new calibrator.

rmackay9, 2015-09-07:
This won't be in Copter-3.3 of course but will go into Copter-3.4.

meee1, 2015-09-24:
this is in beta MP, still working on some ui improvements though.

rmackay9, 2018-07-04:
I think this is long since done so closing.

closed 2018-07-04 by rmackay9