ArduPilot/MissionPlanner #991: Full Parameter List - Changes to values causes a crash of Mission Planner

bionicbone, 2015-09-07:


meee1, 2015-09-07:
please provide a log/video. as this cant be a widespread problem.

bionicbone, 2015-09-08:
Hi Michael,

I've done a video, its available here.
https://www.youtube.com/watch?v=Q3H9JcyoivU&feature=youtu.be

Kev

meee1, 2015-09-08:
can you tell me what version of Microsoft .net you have installed. I have a feeling it will be something to do with the software installed on your pc.

bionicbone, 2015-09-08:
I have everything from 2 upwards I think, with the most recent being v4.5.51209

[image]
[image]

I'll try installing on a different laptop that is a base Win 7

Kev

bionicbone, 2015-09-08:
Hi Michael,

You were correct, it must be a local issue on my laptop as installing V1.3.31 on another laptop, connecting to the same UAV does not have the same problem as I experiance on the original laptop.

Clearly, I'm not the only person to have the trouble as it was mentioned on RCGroups, but also it's not wide spread either othe rise you'd have loads of reports by now.

When I can I will run the VS debugger to see if I can track down the issue, for now I will have to live with it.

Kev

meee1, 2015-09-08:
the problem from what I can see its crashing in the system.windows.forms library, which is a system library, I can only suggest reinstalling the .net framework, V4. and see if this resolves it.

bionicbone, 2015-09-09:
That's easy fixed, I'll  reinstall .net v4 later today.

bionicbone, 2015-09-16:
Hi Michael,

I just downloaded v1.3.32 and the problem seems to be resolved.

This other user also reported the issues being resolved with this release.

http://www.rcgroups.com/forums/showpost.php?p=32698819&postcount=6223

For info I did also install windows 10 so that may have helped or it could have been something you updated.

I'm closing the issue.

Thanks
Kev

closed 2015-09-16 by bionicbone