ArduPilot/MissionPlanner #997: Survey (Grid) - Save trigger method with the camera settings

buckker, 2015-09-11:
Actually the trigger method will not be saved in the survey (grid). So it's inconvenient to change the trigger method for every new auto mission. Very nice would be, if Mission Planner save all camera settings included the trigger method.

meee1, 2015-09-11:
try control-s
and control-o

buckker, 2015-09-12:
I try it, but it doesn't work. With control-s I save the hole mission. If i load it again with conrol-o in trigger method box are the standard values. But my goal is to save all parameters so I can use it on any mission. 

[image]

[image]

meee1, 2015-09-24:
the latest beta MP should save all options.