ArduPilot/MissionPlanner #1000: Reporting of Simple | SuperSimple as MAV_SEVERITY_INFO msg

lvale, 2015-09-12:
This commit went to master, and can/should be reported by the GCS

https://github.com/diydrones/ardupilot/commit/a403e5aef0c9ac351cf2b15418fb0661673e1c91

meee1, 2015-09-24:
by default MP does not show "info" items on the hud, as it causes a bad user experience.

lvale, 2015-09-24:
Michael, as these 2 messages are the first being sent with MAV_SEVERITY_INFO to the best of my grep knowledge :), and as they change the engaged FlightMode behavior I think that any kind of warning on GCS would be useful.

rmackay9, 2015-09-26:
I thought those messages were added so that the OSD could display the info.  I think for sophisticated GCSs we need to add a new message to provide this info to them.  Forcing the GCSs to parse these strings which are only sent once when the state changes means that it's very possible that the GCSs could show incorrect information to the user which will just lead to support calls.