ArduPilot/MissionPlanner #1004: Setting SR0_EXTRA1 on APM3.3 Plane

gibbo9000, 2015-09-15:
I am struggling with setting this parameter on APM3.3 from current version of Mission Planner.  Here is what I find:
a) Connected by USB to PC 
b) Planner > Telemetry Rates > Attitude dropdown changes on screen (from 6 to 10) but doesn't change on APM
c) Full Parameter Tree change of SR0_EXTRA1 makes change (from 6 to 10); Change holds on refresh - but does not reflect back up in the Planner > Telemetry Rates view
d) When disconnect and reconnect SRO_EXTRA1 reverts to 6 in Full Parameter Tree and Telemetry Rates, but SRO_EXTRA2 in Full Parameter Tree changes to 10.
e) Same behaviour with APM Reset (on USB Connect) ticked ot not.  
I am not overly familiar with this architecture yet, and it could well be user error, but this behaviour feels repeatably inconsistent and I am wondering whether there is some cross over of variables in the code.

magicrub, 2015-09-15:
related to https://github.com/diydrones/MissionPlanner/issues/1006

meee1, 2015-09-15:
at what stage are you settings the options? mp can currently have different stream rates per device connected. so depending on when you set it it could change.

magicrub, 2015-09-15:
That's the point. Have the option for Mission Planner to not change at all.

Example: Companion computer+pixhakw boot up. Companion computer uploads mission and sets whatever params it wants to. Launch. 
During the flight you connect to plane via telem2 900MHz to check things out. BOOM, you've changed your SR settings and now the companion computer is receiveing data at the rates Mission planner wants without the companion computer knowing

meee1, 2015-09-15:
what port is the companion computer plugged into, and what port is the telem radio plugged into? as the sr's are independent per port.

magicrub, 2015-09-15:
Really? if so, then we shouldn't need to worry about this. However, I've
seen them change and assumed it was from MP. I'll do some more testing

meee1, 2015-09-15:
mp does rerequest them every 30-45 seconds, but its doesn't directly modify the SR?_ params it uses the request datastream option

gibbo9000, 2015-09-16:
My question is not about whether Mission Planner should change them, it is about consistent behaviour when it does.  For my test above I have APM3.3 Plane plugged via USB direct to laptop running Mission Planner through com3.  No other devices or interferences.  I am also doing it directly after a firmware refresh after reading other posts about difficulties changing.  It is as clean a test as I can get to on the core behaviour.  As a sidebar, in this configuration the SR1 values all change and stay changed in the Parameter Tree.

meee1, 2015-09-17:
so, from you description the issue is that MP is not maintaining it settings on each reload?

gibbo9000, 2015-09-18:
Its behaviour seems to be inconsistent - and hard to pin down.
Using APM Plane 3.3 on Ardupilot direct connect by USB on com3 (and MP 1.3.32) I find:
>  Planner > Telemetry Rates drop downs have no impact on the SR0 flags as reported under the Parameter Tree, eitherreal time or on connect / disconnect:  Are they meant to?
> Changing parameters using the Parameter Tree I can change SR0_Position and it will stay changed through some Disconnect / Connect cycles; but SR0_EXTRA1 always gets written back to its default of 6.
> With APM Reset on USB Connect ticked it is more likely that SRO_Position gets written back to its default of 2, and on a couple of occasions my target SRO_EXTRA1 value in the Planner > Telemetry Rates dropdown has been written to SRO_EXTRA2.
I have tried to quarantine and stabilise what is happeneing but I can't find a repeatable pattern.

To help me, perhaps a couple of questions:
> Should Planner > Telemetry Rates impact SR) values while USB connect?  While Telemetry connect?
> Should APM reset on USB Connect impact this behaviour?

meee1, 2015-09-18:
planner > telem rates will set the sr values for that port, the tick is you need to refresh all params to see the changes. (F5)
apm reset on connect only applies to apm's, not pixhawk.

gibbo9000, 2015-09-18:
Excellent - F5 makes it clearer and more consistent.  It writes the Attitude (SR0_EXTRA1) data to SR0_EXTRA2 as well, but won't write anything bigger than 9 to them!  But on a USB Disconnect / Reconnect it seems to ignore the Planner >  Telemetry Rates dropdowns and writes back the defaults (I assume at point of reconnect?).  So they don't stay set in the EEPROM.  Can you help with the direct Mavlink codes to read them outside Mission Planner - I have been struggling to find them?

gibbo9000, 2015-09-22:
I now have good control of them when running by 3DR telemetry through Mission Planner.  Do you know whether the reset to defaults on connection is driven / triggered by Mission Planner, or is it a function of APM Plane 3.3 that it resets when it gets a new connection?

meee1, 2015-09-24:
the reset on connect from MP only worked for the apm2, and via usb. it has no other function any more

gibbo9000, 2015-09-24:
Thanks for that confirmation. That is consistent with my findings. I can now read the SR0 parameters through Dronekit-Python outside any MP involvement, and can confirm they reset on the APM each time it gets a new connection.  Still working on how to set them consistently through Dronekit!

Saxin, 2017-02-25:
I want to tell a story related to this issue:

I have esp8266 module connected to Telemetry 2 port on Pixhawk, I use it for configuration, logs download and connection to Tower when range permits, recently I've added OSD on same port (OSD connected to Pix TX only of course), I've set SR2 parameters according to MinimOsd-Extra recommendation and here the fun started....

1. I see the parameters "reset" / revert to MP Config -> Planner -> Telemetry rates tab each time I reboot and reconnect.
2. The drop down boxes descriptions aren't 1:1 to Ardupilot wiki -> full parameters list.
3. Not all SRx parameters are there, so I need to toggle between two screens to set them.
4. Can we please have a toggle button not to write to eeprom the values shown on the drop down boxes, I would also expect this to be the default.

Saxin, 2017-03-08:
@meee1 , any news on this issue ?

meee1, 2017-03-08:
MP does not internally change srX_????, it does request the packets at different stream rates, based on the planner config. Any saved setting is characteristic of the autopilot, not mission planner.
also because MP doesn't know what telemetry port it is connecting to, it cant set srX_????. as this would require knowledge that mp does not have.

what is the underlying issue? yes mp is changing the rates, but why isn't MP settings set the same as what you are expecting in the first place? ie set it in MP, and all is fine?

Saxin, 2017-03-08:
I can provide more detailed step by step with screenshots if needed later my evening....

> Any saved setting is characteristic of the autopilot, not mission planner.
also because MP doesn't know what telemetry port it is connecting to, it cant set srX_????. as this would require knowledge that mp does not have.

**Seems like this is exactly what happening**, I would guess the stream rate change is requested on the active port so if USB SR0 change, tel1 - SR1 change and so on....

> what is the underlying issue? 
1. The drop down boxes descriptions aren't 1:1 to Ardupilot wiki -> full parameters list.
2. Not all SRx parameters are there, so I need to toggle between two screens to set them.
3. Suitable for MP or CGS but not suitable for other device connected with only FC TX wire , for example MiniumOSD.
4. This behavior leads to confusion among users, when setting via. full parameter list - one expects the setting to stick....most users won't find this issue....

> yes mp is changing the rates, but why isn't MP settings set the same as what you are expecting in the first place? ie set it in MP, and all is fine?
This is opposed to what written @top of your reply ? _**So does it change or no SRx parameters ?**_

[image]

Off topic a bit - what for the Reset APM on USB request is used ?

This is related - https://github.com/ArduPilot/ardupilot/issues/3239