ArduPilot/MissionPlanner #1007: Survey(grids) Sub-Meter Distance Imput

m11s, 2015-09-15:
Survey(grids)
    right click on flight plan map
    select auto-WP
    Survey(grids)
The input under grid options, distance between lines [m] is limited to a minimum of 1m.
I don't know if this is a simple minimum parameter set somewhere in mission planner, if is an integer vs. float condition, or something else.

Any help with regard to what or where this restriction is would be much appreciated so that I may either fork and fix or submit to have it changed in the main Mission planner.

There is also an option for % overlap in grid options. Which could also solve the problem by reducing the distance between the grid lines by x%. However, this too seems to have a minimum limitation on how close the distance between the lines can be. Which is also approx 1m.

meee1, 2015-09-19:
what is the use case for sub m lines?

m11s, 2015-09-20:
The use would be for a rover to fully cover a defined area (like a
lawnmower would). The rover in most cases will be significantly less than
1m wide.

Thank you for your help.

meee1, 2015-09-20:
how low do you want to go? 0.3?

m11s, 2015-09-20:
That might work. But it would limit my rover to only certain sizes. I was
hoping cm or decimeter.  .3 would be almost 1ft which would be fine as a
minimum but increments of .3 would probably be to large a jump. If the
overlap function would also allow a smaller minimum, fine tuning could be
done there instead or in addition to the path separation.

meee1, 2015-09-24:
this should be in the latest beta

closed 2015-09-24 by meee1

Jarodirvin, 2017-01-04:
I am not able to get sub 1m spacing on simple grid. If this was implemented over a year ago why was it removed?

Jarodirvin, 2017-01-05:
I am referring to spacing between lines. The intent is to have a small Rover (8" wide) that can cover an entire area.

meee1, 2017-01-05:
how small do you plan on going? current version min is 0.3 m ie ~1ft

Jarodirvin, 2017-01-05:
0.3 should work. I must be doing something wrong then. In mission planner, create polygon, auto WP, simple grid, distance between lines - entering anything below 1 will cause it to switch to one.

meee1, 2017-01-05:
your issue is simple grid. the changes above are for survey grid, not simple grid.

zacharyDez, 2019-09-17:
@m11s  have you been able to execute the mission with sub-meter spacing? Have been trying to execute similar mission with an R1 rover. Mission at 3m work but unexpected behavior arises with submeter/meter spacing. Did you have issues with the number of waypoints being written as well?

Jarodirvin, 2019-09-17:
I was not sucessfull, the robot kept trying to find the waypoints, but fell apart before i made anything usefull, i was not ever able to load enough waypoints to program what i needed, then other projects distracted me and have not messed with it since. Wish i could have helped you.

Sent from my Verizon, Samsung Galaxy smartphone

nichord, 2023-06-01:
I know this is an old thread, but wanted to reopen this as we have gotten RTK solutions. I have a robotic mower with around 13" cut width, but would like to have some overlap ability. Is there a reason for the .3m minimum, or can we drop that down to something like .1 (or whatever the code would work with for min). With sub cm accuracy (while not fully achievable at all times, would allow some overlap for small vehicles like a robot mower to make sure all grass is cut. Thoughts?