ArduPilot/MissionPlanner #1009: Load .bin logs directly

AndKe, 2015-09-19:
it's time to load .bin logs directly, instead of needing manual conversion to .log first :)

meee1, 2015-09-24:
done, however there is an underlying issue with mono not supporting datagrid virtual mode. so on mono you will never be able to open large logs.
i have seen .bin files up to 600mb in size, which on windows this is possible.

AndKe, 2015-09-25:
Thanks, I know you are very locked in with .NET, but do you have any thought towards make MP a multiplatform application ? - like using Qt (or anything else)?    I do know it's a lot of work and I don't expect a "yes next week"  ,but is it something you ever think about ?

-mono is becoming better with each version as MS opened the code, the alpha mono is great, but MP is one of the apps that can be crashed in very many ways while run in mono,  - are you interested in more crash reports from mono crashes ?