ArduPilot/MissionPlanner #1010: Survey Grid improvements

Jman841, 2015-09-20:
Two things that would be helpful with planning mapping missions:

1. Being able to define a large area you want to map, set the maximum flight time of the airframe, and have survey grid automatically break up the area into different flight segments with X overlap between flights. This would make planning large area's much simpler and you could simply save each flight as a different waypoint file. When you finish the first flight, land, change batteries, load the second waypoint file, and fly again. No guess work on exactly where you left off or how much overlap you need between projects.

2. Use terrain data to show what the minimum overlap would be. We had this issue recently when mapping very mountainous terrain. All the estimations for overlap and side-lap are based on the altitude of the home beacon, however, if mapping a mountainous area, the overlap and side lap at higher elevations than where the survey grid made the estimations will be considerably less and can effect the quality of the orthomosaic. If there could be a "Minimum overlap" indicator based on the elevation of the area you are mapping, it could help set the proper parameters. Or, even more complex, have it change the overlap timing based on the terrain? Side lap would be difficult to adjust, but, overlap could be simple.

iskess, 2015-09-20:
+1!!
Please see #573

meee1, 2015-09-20:
for 1. mp currently implements a resume mission button, that was pertly designed to fit this situation.

you plan the mission as one, when battery rtl is called, mp stores the last wp it used. you land, switch battery, and then click the resume mission button, and it will auto populate the mission with a do jump to get the autopilot back onto the previous point.

Jman841, 2015-09-20:
Very interesting. Is there a way we could "pause" the mission other than a low battery trigger? I personally don't like flying to the low battery alarm on purpose. Any way to initiate it manually?

meee1, 2015-09-20:
simply doing an rtl would archive that.

Jman841, 2015-09-20:
Very interesting, I must of missed this addition. Thank's for the tip. I'll search the wiki to read the details about it.

So it basically cuts the mission, then when you push the button it adds a takeoff parameter based on the original one?

meee1, 2015-09-20:
correct.

Jman841, 2015-09-20:
One more question with that, how does that work with the geotagging/logging. I assume there will be 1 .tlog for all of it, but, 3 data flashlogs?

Is the .tlog just as good as using data flashlogs for geotagging with mission planner?

meee1, 2015-09-20:
so long as you don't disconnect mp, yes one tlog.
there would be 3 df logs, I havnt realy thought about how you could deal with that.

how do you normaly process your logs? using cam messages? or time offset?

Jman841, 2015-09-20:
I have tried both but have not determined what works best. With Cam messages, if the pictures are off by 1 compared to the messages, which is quite common to either have 1 extra picture or 1 less picture, it simply throws an error and doesnt work. With time offset I do not fully understand the algorithms behind it to best understand which would be more accurate. The third option is to just load the .log into pix4d and let it use their algorithms to determine which seems to be the simplest.

What is the most accurate way to add geotags? 

If you have 1 less or 1 more picture than cam messages, can it still work properly if you delete a picture, or use a duplicate picture to "fool" it into working? Or could this possibly mess it all up?

meee1, 2015-09-20:
if you have 1 more, or one less, use time offset, but tick the use cam message box, under time offset.
this will scan the df log for just cam messages, and make a best guess based on just the cam messages vs then entire log's gps points.

Jman841, 2015-09-20:
ok, thank you. 

Is there any main difference in using the .tlog vs the dataflash logs?

Would there be a way to integrate multiple dataflash logs for analysis into the Geotagging tool on mission planner? That may solve this issue, although, it would be tough to determine which pictures are from which flight if the naming convention of the camera was not perfect.

meee1, 2015-09-20:
DF will normally have more data points to work with. ie 5hz gps data, or the equivalent cam message data.
whereas tlog will have normally 2hz gps data.

meee1, 2015-09-20:
,oh and one more thing, this is more a FYI. in the grid screen try control-S to save. and Control-O to open a save.

Jman841, 2015-09-21:
Ok, i'll give it a try. That will stop it from defaulting to 5 m/s for the flight correct? 5 m/s will stall my plane if I forget to change it.

Jman841, 2015-09-21:
I just played around with the restart mission button on the bench (Connected through the USB) to see how it works. It loaded the mission properly, however, the first way point is a Waypoint and not a takeoff command. Is this correct?

meee1, 2015-09-23:
the current resume was designed for copter, ive updated it to support plane

Jman841, 2015-09-23:
Ok, thank you. I was testing it yesterday and nothing was working right. It kept putting it in guided mode and putting the elevator like it wanted to nose down dive. That would explain it.

A few other possibilities for the survey grid improvement.

1. Is there a way it could save the default airspeed based the last time used? Or if it recognizes Arduplane vs. Arducopter use different values? The default 5 m/s would be detrimental on a plane if someone forgot to change it.

2. For survey grid V1, would it be possible to change the polygon inside the Survey Grid? Survey grid V2 does this, but, with V1, if you realize you choose too much or need to make changes, the only way to do it is to exit out, delete the polygon, and remake another one. The polygon tool itself is not exactly the easiest to use as half the time you end up making a new point on accident when trying to move a point.

3. Already mentioned before, but, using the terrain data to account for the amount of pictures it needs to take in specific areas.

Thank you for your work on this. It is an amazing tool!

meee1, 2015-09-24:
ive made some mods to the grid tool to do split missions. at the moment it works very simply. it divides up the total wps and splits it.
adding a takeoff and rtl/land at the start/end of each segment.

Jman841, 2015-09-24:
Excellent. I will test it soon. Thank you again!

IronDome2, 2016-03-06:
Dear all

I'm very delighted to read these comments as they are all relevant to my situation too.
First I would agree on the default speed 5 is quite scary, I use Km/hr and my plane does between 50 and 60. so I simply add a zero in there. However, I only use it fr estimating my flight time for my mission, so I can't really crash my bird. But would be nice to see a much more reasonable number in there.

With regards to breaking up the mission area, what has been suggested is good, by in my case I fly a fixed wing plane which is recovered with a parachute. So part of my mission, is to  establish recovery line which will have about four way points. Therefore if i had to break my mission as suggestion here, it would be quite tedious job to setup my recovery line for every flight. what I normally tend to do for large area to be covered, I would create separate recovery line and have it saved. So every end of of mission, I  simply remove the existing mission plan and load up recovery line and. But again the challenge is that, MP  does not show some kind of foot print on the already covered area so that planning for the next block should be easier as you would see that foot print.

Also, just as Jman has pointed out the map tools are quite unfriendly. Doing the polygon in most case I would make mistakes, distance measuring tool could be better, map grid also could be improved and a scale bar could be introduced. 

When loading KML overlay, there is always a whitish fill, how do I get rid of that? For now in Google earth I just create polyline instead of polygon, but then I can't get area calcs with polyline. The other think i find pretty difficult is to change my home while the aircraft is not connected and powered. For now I have to drag the H way-point to where I want, but is there a better way? i  totally agree that if we could have elevation data to assist in determining footprint coverage, it would be very useful. I have been spoiled by eBee's  eMotion software which has all these tools, off course it is commercial so they have more resources to do the development work. 

My other big challenge is that, i want to stop Camera trigger just before the plane arrived to the way point and starts to roll and start trigger when the plane is on the new line and is level. Presently I use SRT_CAM_TRIGG_DIST which does not care whether the plane is level or not. Any ideas?

Thanks for the wonderful tool.

bortek, 2016-04-12:
@meee1 Today I had built a simple mission using Survery v1. I split it into 2 segments using a check box in the Survey window. That generated a waypoint list starting with 2 Jump commands. 
1. DO_JUMP-> WP3
2. DO_JUMP-> WP22
3.TAKEOFF
......
<CUT HERE>
SOME WAYPOINTS IN BETWEEN
<CUT HERE>
...
21. RTL
22. TAKEOFF
......
<CUT HERE>
......
37. RTL

When I flown the mission the first section went well and when it was completed it Pixhawk executed WPT#21 which was RTL and landed. The motors slowed down a little but never stopped. Then after 2 seconds it did a takeoff and continued with second section of the mission.

I did not have any mission planner connected via telemetry. I just started the mission with my radion button going into Auto mode.

My first question is when my hexacopter lands after the first section how do I pause the mission to change the batteries and then continue. Do I need to do it with the mission planner connected via Mavlink ?

Another question is why is WPT#2 needed which is JUMP -> WPT#22. This will never be executed as far as I can see.

Last question. Next to the Takeoff command is in the first column number 20 (?) and in the last column Alt (30). Whats the first column for TAKEOFF, I cannot find it's meaning on Wiki.

meee1, 2016-04-12:
@bortek I was not aware it did that. the takeoff after the rtl.
as for the no method to get to wpt2, that is correct, and by design, and will always require user intervention. you need to set the wp to 2, and then  switch to auto to make the second half of the mission work.

bortek, 2016-04-13:
You mean i need to manually intervene and set wpt2 and then run auto to execute second half of the mission? 

This is a little bit of work for automated mission. I thought the purpose of split function was in the first place  to replace the batteries. What is its actually purpose?

IronDome2, 2016-04-13:
Very good post (original).
My additional questions are:-

1. Is it possible to have camera shutter audio as photos are taken?
2. Is possible to show photographed area foot print as mapping progresses?

Thanks in advance.

meee1, 2016-04-13:
for 2, please check https://github.com/ArduPilot/MissionPlanner/issues/1210#issuecomment-209554671

bquijano, 2017-02-01:
@bortek how it went? did you manage to swap batteries and continue the mission?
I'm looking for the best solution to that problem. I did't test any  options so far.
Cheers, 
Bernard

bortek, 2017-02-03:
@bquijano I did not quite got it tested actually as other things poped-up you know :) Please report your results here if you get it tested. Thanks!