ArduPilot/MissionPlanner #1013: Flap Control through Mission Planner

Jman841, 2015-09-23:
One issue I have is, if I don't have my Tx on or don't plan to use my Tx for a flight. the Flaps on my plane automatically default to full. Within Mission planner there is no way for me to bring the flaps up without getting my Tx, putting them up, then turning the Tx off.

Could there be a way to deploy and retract flaps from Mission planner? Might have to add a parameter to select the Flap Channel and tell it which PWM Value is flaps up, vs. Flaps down. But it would be a great feature, especially if flying through a joystick and wanting to deploy the flaps for landing or something.

Thank you

meee1, 2015-09-23:
have you looked at using the servo tab?

Jman841, 2015-09-24:
Yes, it does not work. For some reason if I toggle the flap servos it flicks and goes right back down to full deflection as without the tx on I believe the Rx is putting out the PWM for full flaps. If mission planner could have a "Flaps Up" or "Flaps Down" button or even a Toggle Flaps button that could override the Flaps channel it should fix the issue.

meee1, 2015-09-24:
the problem is which one is master? MP or the radio. you need to make this call yourself. as only one can be master.

Jman841, 2015-09-24:
The radio is powered off, however, mission planner has no way to control the flaps.

Even if the radio is powered on or off I can still change flight modes and do all other functions, the only function I can not do without the Tx currently is operate the flaps. For some reason when my Tx is off, the default position of the flaps is full down. The Issue is I can not change this behavior in mission planner.

meee1, 2015-09-24:
i think you may need to bring this issue up for arduplane, not MP, as currently there is no method this can with via.

Jman841, 2015-09-24:
Ok, thank you.

GrahamDyer, 2015-10-03:
You can control flaps automatically: http://plane.ardupilot.com/wiki/automatic-flaps/
or you can use the servo tab in MP to control them, you MUST however manually set their min/max PWM values (or end points) in the MP otherwise it will not work (eg. RC6_MIN/MAX). The high/low values will control how much to move them when activated.