ArduPilot/MissionPlanner #1014: RAWX to RINEX

ThorstenDP, 2015-09-23:
Hi all,
when enabling RAWX logging Mission Planner generates an *.obs file when downloading the logs. However, it only induces some header information. From some discussion over at the ardupilot repository, I thought that RAWX->RINEX is already implemented - but not really sure. 
So, it is some bug? If it is not implemented so far, are there any plans to do so?
Thanks a lot and best regards,
Thorsten

meee1, 2015-09-23:
it was working, what gps are you using? and do you have a log I could grab to test against.

ThorstenDP, 2015-09-24:
Thanks for having a look! 
I am using an M8T and as far as I can see all required data (GRXH and GRXS) is logged. Hence, I assume all is correct on the AC3.3 side.
Here is a link to a log: https://dev.bitmapping.de/index.php/s/vkRuZUYfGqiTYIN
What might be the problem is that there is a GRAW line in the  FMT messages. Maybe Mission Planner is looking for further but missing GRAW data and ignores the GRXH and GRXS data.
Best regards,
Thorsten

meee1, 2015-09-24:
i just did a conversion, and dont see the issue.

what was the problem you where having?

what version MP are you using?

meee1, 2015-09-24:
[image]

ThorstenDP, 2015-09-24:
I am using the latest beta version. 
With all files I tried so far there was only an *.obs file with the following content:

     3.02           OBSERVATION DATA    M: Mixed            RINEX VERSION / TYPE
                                                            MARKER NAME         
                                                            MARKER NUMBER       
                                                            MARKER TYPE         
                                                            OBSERVER / AGENCY   
                                                            REC # / TYPE / VERS 
                                                            ANT # / TYPE        
        0.0000        0.0000        0.0000                  APPROX POSITION XYZ 
        0.0000        0.0000        0.0000                  ANTENNA: DELTA H/E/N
G    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES 
S    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES 
R    4 C1C L1C D1C S1C                                      SYS / # / OBS TYPES 
E    4 C1C L1C D1C S1C                  

I'll check , vs .

ThorstenDP, 2015-09-24:
Yes, ist is/was a , vs . problem.
I am still having some problems in RTKLIB, but I will have a closer look at that later today.
Thanks a lot for your help!

meee1, 2015-09-24:
I did see that coming
so did this
https://github.com/diydrones/MissionPlanner/commit/84043d2200aa9068437f5479da547da98126fb63

closed 2015-11-25 by meee1