ArduPilot/MissionPlanner #1015: New RSSI parameters in ArduPlane 3.4.0 have no help

StewLG, 2015-09-25:
The new RSSI parameters in ArduPlane 3.4.0 have no help in Mission Planner - I tried release and beta, specifically Mission Planner 1.1.5745.29653.

[image]

If I'm missing something about the way releases work please set me straight.

meee1, 2015-09-25:
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_RSSI/AP_RSSI.cpp
has malformed param names in it.
"RSSI_" is gathered from the main paramerter.cpp, however in the linked file it also has RSSI_*.
so either the main parameter.cpp needs to be changed to "" or the linked file needs to remove the "RSSI_"

meee1, 2015-09-25:
eg http://plane.ardupilot.com/wiki/arduplane-parameters/#rssi__parameters

closed 2015-09-25 by meee1

StewLG, 2015-09-26:
This has now been fixed in trunk. Thanks for the explanation.