ArduPilot/MissionPlanner #1020: Easy search for Full Parameter List

GrahamDyer, 2015-10-03:
It would be very convenient to have a persistent search box in the right hand side-bar which populated as soon as anything was typed after loading, rather than having to click "Find" or CTRL-F (which first brings up the temp screen) and then typing.

This wouldn't interfere which parameter editing as it could/would lose focus as soon as any parameter was clicked. 

With so many parameters, unless one knows the name exactly in which case one can scroll to it, the way to find something is to search and this would just make it that much easier and faster.

iskess, 2015-10-03:
+1, this would be a huge help!

It would be really nice to have a custom parameter page where we could arrange our own list of frequently used parameters. Or the ability to hide parameters that we will never look at, along with a "show all" button to reveal all hidden items. 
As the parameter list continues to grow, it is slowing down the workflow in the field.

GrahamDyer, 2015-11-08:
Works very, very well on the beta, impressive, thank you! 
Another small suggestion: it could have a small "clear" button to clear the box of text to go quickly back to the full list.

GrahamDyer, 2015-12-24:
Works amazingly well, thank you Michael!

closed 2015-12-24 by GrahamDyer

meee1, 2015-12-24:
good to hear

GrahamDyer, 2017-05-29:
Best addition ever :)