ArduPilot/MissionPlanner #1022: Flight Plane: allow specifying altitude type as relative, absolute or terrain

rmackay9, 2015-10-05:
Copter-3.4 will include support for mission items with altitudes specified as relative (what we currently support), absolute alt above sea level and altitude above terrain.

[image]

I think we should add a drop-down perhaps which allows selecting the frame which is saved to the third column of the .txt files.
0 = absolute alt above sea level
3 = relative alt above home
10 = alt above terrain

meee1, 2015-10-06:
currently mp forces the entire mission to a specific type. its not selectable per wp.

rmackay9, 2015-10-06:
I'm unsure whether we need a per-command setting or whether setting it for an entire mission is enough.  I suspect it won't be long after terrain following is implemented that people will ask that they be allowed to mix alt types but I'm happy to at least wait until that request comes in.

In the vehicle code we will implement it so they can be mixed.

iskess, 2015-10-06:
I requested the ability to select altitude reference for each WP a year ago #566, (see post 2 and 3). Tridge provided this functionality in Mavproxy.

meee1, 2015-10-06:
and I still don't plan on changing it at this point in time, as I believe it will create user confusion.

naktinis, 2016-05-03:
Can we consider this closed after 3547b4ae2139589b25192b16a04acbe6fe996aab? Issue description does not explicitly ask for per-command altitude reference.

meee1, 2016-05-03:
@naktinis sounds fair to me. that  and randy has tested this

closed 2016-05-03 by meee1