ArduPilot/MissionPlanner #1026: no georeference possible

Basti77, 2015-10-07:
The georeference of the images dont work in in the current Mission Planner version.

i meen the hidden Controll+f Option.

The bin file is converted to the log Format, but it is no position data stored in relation to the camera trigger.

in earlier versions, this has worked.

meee1, 2015-10-07:
please provide a log, so I can check against

Basti77, 2015-10-07:
Hello Michael.

http://www.droneshare.com/mission/152575

Where should I upload the Bin File? Zip files are not accepted here.

Greeting Sebastian

Basti77, 2015-10-08:
I've found the problem.
in the georeferencing tool is missing the console.
You get no feedback for debugging.

My problem was, the camera has 2 more images recorded as than planned.

Would it be possible to install the console back in the current version?

meee1, 2015-10-11:
to get the console, just goto help an tick the console box.

closed 2015-10-11 by meee1

Basti77, 2015-10-12:
I mean this area with debug console.
[image]
[image]