ArduPilot/MissionPlanner #1036: Auto analysis, GPX creation and upload to Droneshare not working

GrahamDyer, 2015-10-16:
Currently any of the logs created by Copter v3.3 on my tricopter do not work or fail in the following errors:

Auto analysis fails with:
[image]
and:
[image].

GPX creation works but the file has an error:
[image]

Upload to Droneshare fails with:
[image]

foobarbecue, 2015-10-18:
n.b. I get "invalid location" from apmplanner2 as well. Droneshare server side is not open source so pff

meee1, 2015-10-20:
could you upload the gpx that was created? just so I can see whats invalid in it?

GrahamDyer, 2015-10-21:
Delete the ".gif" of the filename
[image]

meee1, 2015-10-21:
I cant see anything wrong with the gpx file

GrahamDyer, 2015-10-21:
Went an tried it on a different viewer and it works fine, don't know why Garmin Basecamp reports an unknown error on that file.

closed 2015-11-25 by meee1