ArduPilot/MissionPlanner #1037: Minimum WP_RADIUS Achievement

m11s, 2015-10-17:
In the flight plan tab under way points at the bottom there is an option to set the Way point Radius.
The minimum allowed input is 1m.
With new RTK GPS accessories coming to market for APM this needs to be in increments of 1cm(.01m) to compliment the accuracy that devices like the Piksi and others can now provide.

Possible source in XML UI:
https://github.com/diydrones/MissionPlanner/blob/e02a8f51b2a0ba4f5e0ad5202c922e1ad49a75a2/GCSViews/ConfigurationView/ConfigArdurover.resx#L810

Possible source:
https://github.com/diydrones/MissionPlanner/blob/1121493593d36976c550cfe4bbf747c4217daf1b/GCSViews/FlightPlanner.cs#L2084

Are there any inherent limitations using sub-meter GPS coordinates in autonomous mode?

If anyone knows of any other refinements that need to be made for cm level operation please mention them here also.

meee1, 2015-10-20:
this needs to change in ardupilot first. as it is stored as an integer.
I will close this until it is possible in the main code.

closed 2015-10-20 by meee1

m11s, 2015-10-21:
These parameters seem to show some support for cm waypoints in Ardupilot.
(WPNAV_WP_RADIUS)  and  (_wp_radius_cm)  located in the link below in 

ardupilot/libraries/AC_WPNav/AC_WPNav.cpp

https://github.com/diydrones/ardupilot/blob/3d7ec3704d88053ae03b6e01077946ec2084af99/libraries/AC_WPNav/AC_WPNav.cpp#L26

Is there a way to use these or should I raise this issue in diydrones/ardupilot?

meee1, 2015-10-21:
agh, in the issue you referenced WP_RADIUS, and that all I had to go on.for copter, and the way the leash works, there is no point in making it sub m anyway.

m11s, 2015-10-21:
This would be used on a Rover. 
Could 

@Param: RADIUS
@DisplayName: Waypoint Radius

be used or implemented in Mission Planner if the minimum were lowered below 100cm?